KinesisCorporation/Adv360-Pro-ZMK #208: non-detectable on FreeBSD

Dmitry-N-Medvedev, 2023-08-16:
good morning!

First of all I must apologize if this is not the right place to report my problem.

**The Context**: my adv360pro is paired to a macbook pro, a dell windows laptop. My FreeBSD desktop can see the macbook, the windows laptop, and an apple keyboard.

**The Problem**: my FreeBSD desktop cannot see my adv360pro.

I don't understand this -- how the adv360pro can hide from a bluetooth scan?

I need your help. Desperately.

best regards,
Dmitry

ReFil, 2023-08-16:
Does your desktop support bluetooth LE? not all bluetooth adapters and drivers support it

Dmitry-N-Medvedev, 2023-08-16:
hi,

thank you for coming back to me!

FreeBSD is supposed to support BLE.

I am not sure if macbook pro or apple keyboard speak BLE, but if they do, I
can then confirm that my FreeBSD desktop can talk BLE, since I can see,
say, apple keyboard via the following command:

*doas hccontrol -n ubt0hci inquiry*

Inquiry result #0
BD_ADDR: 10:94:bb:b1:0a:d8
Page Scan Rep. Mode: 0x1
Page Scan Period Mode: 0x2
Page Scan Mode: 00
Class: 00:25:40
Clock offset: 0x58f6

Dmitry-N-Medvedev, 2023-08-16:
> The apple keyboard doesn't use BLE as far as i know

ah, okay.

anyways, I cannot explain why the FreeBSD desktop sees an apple keyboard
and does not see the adv360pro.
taking into consideration that the adv360pro is paired with macbook pro.

I must be misunderstanding something very basic here.

ReFil, 2023-08-16:
The macbook pro supports regular bluetooth and bluetooth LE, The freeBSD desktop only supports regular bluetooth but the adv360pro only supports bluetooth LE

Dmitry-N-Medvedev, 2023-08-16:
aha, okay, now it is more clear.

the hccontrol program has a bunch of commands that affect BLE settings:

le_enable
le_read_local_supported_features
le_read_supported_states
le_set_scan_response
le_set_scan_enable
le_set_scan_param
le_set_advertising_enable
le_read_advertising_channel_tx_power
le_set_advertising_param
le_set_advertising_data
le_read_buffer_size
le_scan
le_read_white_list_size
le_clear_white_list
le_add_device_to_white_list
le_remove_device_from_white_list
le_connect
le_read_channel_map
le_read_remote_features
le_rand

since I have never configured bluetooth/BLE before, maybe you know the
proper BLE settings the adv360pro expects to see?
I would then try to apply those via the said commands and see if there is
any change.

best regards,
Dmitry

Dmitry-N-Medvedev, 2023-08-17:
okay. at this moment what is really missing is the expected workflow of establishing connection to the adv360pro from the perspective of BLE. I can confirm FreeBSD can see the keyboard via BLE, but returns only its random address. Which is weird, because MacOS shows the same f9:1a:f2:c6:b5:b9 address of the keyboard all the time.

It is obvious that I lack BLE related knowledge, hence I kindly ask Kinesis Software engineers to assist me with proper workflow ( apart from the tools ) -- I just need to learn which steps are involved and what data I should see at each step.

best regards,
Dmitry

ReFil, 2023-08-17:
Does FreeBSD support Bluetooth LE resolvable private addresses?

Dmitry-N-Medvedev, 2023-08-17:
> Does FreeBSD support Bluetooth LE resolvable private addresses?

yes.

```
dmitry@z420 ~ $ doas hccontrol read_neighbor_cache
T BD_ADDR Features Clock offset Page scan Rep. scan
R 4d:7a:60:12:c6:7e 00 00 00 00 00 00 00 00 0 0xc3 0
Complete local name: Adv360 Pro
Appearance: c1 03
Flags: 06
Incomplete list of service class UUIDs (16-bit): 12 18 0f 18
```

Dmitry-N-Medvedev, 2023-08-21:
the question is: what is the sequence of steps that would lead to the said keyboard connected to the freeBSD machine?

@ReFil please share your knowledge

PS: iOS can do this, MacOS can do this, event Windows can do this. I don't believe that freeBSD cannot. I just need to learn the procedure in plain English and will try to automate it with a shell script.

ReFil, 2023-08-22:
the R is probably resolvable private address. You need to initiate pairing with the keyboard, the Adv360 pro uses simple pairing, no passkey or anything, just BLE bonding. Ensure the profile on the keyboard is empty or cleared

Dmitry-N-Medvedev, 2023-08-22:
> the R is probably resolvable private address. You need to initiate pairing with the keyboard, the Adv360 pro uses simple pairing, no passkey or anything, just BLE bonding. Ensure the profile on the keyboard is empty or cleared

good morning @ReFil. Thank you for your message!

I am educating myself regarding the BLE and found this article: https://technotes.kynetics.com/2018/BLE_Pairing_and_bonding/

so, a quick question: which version of the BLE does the keyboard use?

best regards,
Dmitry

ReFil, 2023-08-22:
LE Secure connection "Just works" pairing

Dmitry-N-Medvedev, 2023-08-22:
well... BLE is way to difficult to wrap my head around. Closing this.

closed 2023-08-22 by Dmitry-N-Medvedev