KinesisCorporation/Adv360-Pro-ZMK #213: The "r" key (Dvorak) will double type the character

mertzjames, 2023-08-28:
I've noticed recently that my keyboard will sometime double type the `r` character using the custom dvorak layout (the `o` character in QWERTY).  It appears to only be this specific character on all layers I've defined including the numpad.

Here is an example output of the keyboard of just simply typing `r<space>r<space>` repeatedly.

```
r r r r r r rr r rr r r r r r r r r r r r r r r r r r rr r r r r r r r r r r r r r r r r r r r r r r r r r r r r r r r r r r r r r r r r r r r r r r r rr r r r r r r r r r r r r r r r r r rr rr r r r r r r r r r r r r r r r rr r r rr r r r r r r r r r rr r r r r r r r r rr rr r r r r r r r r r r r r r r r 
```

As you can see there doesn't appear to be any consistency to when the action happens.  I'm curious if there's a firmware issue or if there's something that I can do to debug this.

My specific keyboard settings can be found here: https://github.com/mertzjames/Adv360-Pro-ZMK-Dvorak

KinesisCorporation, 2023-08-29:
Hi James,

That sounds more like dirt/debris in the key switch then a firmware issue. Try some canned air. Would be worth while reverting to a different version of firmware also and seeing if the problem persists. 

If it does, please submit a trouble ticket to us and we can issue an RMA

https://kinesis-ergo.com/support/contact-a-technician/

ReFil, 2023-09-02:
This definitely sounds like a hardware issue, I'm aware that certain mechanical switches are bouncy enough to cause problems with the stock debounce settings (Boba U4 switches) but this would only be a concern if you got a signature series 360 pro from upgrade with those switches fitted and it wouldn't be limited to only one key. Closing as unplanned

closed 2023-09-02 by ReFil