KinesisCorporation/Adv360-Pro-ZMK #216: Could not find a configuration file for package "Zephyr-sdk" that is compatible with requested version "0.15"

tiziano88, 2023-09-01:
When running `make` locally, I consistently get the following error:

```
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
CMake Error at /app/zephyr/cmake/modules/FindZephyr-sdk.cmake:63 (find_package):
  Could not find a configuration file for package "Zephyr-sdk" that is
  compatible with requested version "0.15".

  The following configuration files were considered but not accepted:

    /opt/zephyr-sdk-0.13.2/cmake/Zephyr-sdkConfig.cmake, version: 0.13.2

Call Stack (most recent call first):
  /app/zephyr/cmake/modules/FindHostTools.cmake:53 (find_package)
  /app/zephyr/cmake/modules/dts.cmake:8 (find_package)
  /app/zephyr/cmake/modules/zephyr_default.cmake:108 (include)
  /app/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /app/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:15 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/app/build/left -GNinja -DBOARD=adv360_left -DZMK_CONFIG=/app/config -S/app/zmk/app
make: *** [Makefile:16: all] Error 1
```

Now, I suspect this is due to the fact that the base Docker image is not pinned at all:

https://github.com/KinesisCorporation/Adv360-Pro-ZMK/blob/2d369451ae2a1326f64c37221802e1f4079b590a/Dockerfile#L1

Therefore I probably still have an old version of that cached locally.

I strongly recommend pinning the Docker image (and every other dependency) to a specific digest, to ensure that the build works identically for everyone.

tiziano88, 2023-09-01:
If anyone comes across this, the ~fix~ workaround is to run

```
docker pull docker.io/zmkfirmware/zmk-build-arm:stable
```

ReFil, 2023-09-02:
`make clean` will delete the docker image allowing `make` to pull down a fresh one, which saves messing around with docker directly

ReFil, 2023-09-06:
Closing issue as solved

closed 2023-09-06 by ReFil