KinesisCorporation/Adv360-Pro-ZMK #217: Documenting key positions for ZMK use

debashisbiswas, 2023-09-02:
Some settings in ZMK, such as combos, require the layout's key positions. I've figured out the key positions for my own config, and would like to document them in this repo somehow.

Here's what I ended up with my in config.

Where's the best place to put this information for other users?

```
#define KEYS_L   0  1  2  3  4  5  6 \
                14 15 16 17 18 19 20 \
                28 29 30 31 32 33 34 \
                46 47 48 49 50 51    \
                60 61 62 63 64

#define THUMBS_LEFT     35 36 \
                           52 \
                     65 66 67 \

#define KEYS_R   7  8  9 10 11 12 13 \
                21 22 23 24 25 26 27 \
                39 40 41 42 43 44 45 \
                   54 55 56 57 58 59 \
                      71 72 73 74 75

#define THUMBS_RIGHT 37 38    \
                     53       \
                     68 69 70 \
```

ReFil, 2023-09-02:
What you could do is create a KLE image with thr rows, columns and key positions annotated in a similar fashion to this: https://github.com/PolarityWorks/zmk-config-ckp/raw/main/rowcolmap.png

I'd be happy to add such a file to the repository, i think it would be very useful

LeonB, 2023-09-04:
@debashisbiswas Thanks! Was just looking for the key positions today in combination with the tri-state zin feature.

debashisbiswas, 2023-09-04:
Here's an image I generated using https://github.com/caksoylar/keymap-drawer.

[image]

It may also be useful to include these in plain text to make it easier for users to copy into their configs.

KinesisCorporation, 2023-09-05:
Awesome, thanks!

PR #222 (2023-09-07): Readme rewrite, add key position documentation
Resolves #217
closed 2023-09-07 by ReFil