KinesisCorporation/Adv360-Pro-ZMK #224: New Keyboard Today. Cannot connect to MacBook Pro

PetulantPenguin, 2023-09-11:
I just got the ADV360.

One Adv360 appears in the bluetooth New Devices list.
But, it doesn't connect. I tried restarting the Macbook. But, no change.

Apple M2 Pro
Venture 13.5.2

ReFil, 2023-09-11:
Have you used the bluetooth clear button? (mod+ right windows)

ReFil, 2023-09-13:
Just following up on this, are you still encountering issues?

PetulantPenguin, 2023-09-13:
I did not try that yet. I will try shortly.

I remapped some keys and restarted the keyboard a couple times yesterday and the day before. Bluetooth still didn't connect after that.

ReFil, 2023-09-13:
Also try connecting it to a smartphone or other device to ensure its functioning correctly

PetulantPenguin, 2023-09-13:
I connected it to my iphone and it worked.
on my macbook, mostly i see 1 ADvPro in the bluetooth settigns. sometimes (rarely) i see 2 of them.
I click connect next to the device. Sometimes it flips to Disconnect briefly, but then goes back to connect as if nothing had happened. and the keyboard always remains in the new devices section of the bluetooth window

ReFil, 2023-09-13:
Press mod+2, ensure the profile light is fast flashing and try pairing it to your Mac again

PetulantPenguin, 2023-09-13:
Fast flashing. Same deal.

ReFil, 2023-09-13:
https://github.com/KinesisCorporation/Adv360-Pro-ZMK/blob/V3.0/config/boards/arm/adv360/adv360_left_defconfig#L52-L57 try removing these lines on your repo

KinesisCorporation, 2023-09-13:
If the Profile LED is fast flashing it sounds like the keyboard is ready to be paired. 

Is there an option to delete/forget/remove the device from you computers Bluetooth menu?

> I just got the ADV360.
> 
> One Adv360 appears in the bluetooth New Devices list. But, it doesn't connect. I tried restarting the Macbook. But, no change.
> 
> Apple M2 Pro Venture 13.5.2

Can you provide a screen shot of your Bluetooth menu. On our macOS Ventura MBP we have "My Devices" and "Nearby Devices", but nothing called "New Devices"

PetulantPenguin, 2023-09-13:
That’s what it is. I’m not at the house right now, but I will send you a screenshot soon.

KinesisCorporation, 2023-09-18:
@PetulantPenguin 

Any update?

KinesisCorporation, 2023-09-28:
@PetulantPenguin Bump

PetulantPenguin, 2023-10-01:
sorry. i took a trip across the country. will be back with time to try it tomorrow.

thanks for keeping with me.

KinesisCorporation, 2023-10-20:
We will close this issue unless we hear back soon. Thanks!

ReFil, 2023-10-25:
Hello, just following up on this, please try the latest update as it has several improvements to help with host connections, if we dont hear back within 48 hours we will consider this issue stale and close as such

closed 2023-10-27 by ReFil