KinesisCorporation/Adv360-Pro-ZMK #227: Kinesis Advantage 360 Pro Left Module Firmware Flashing Issue - [the Left is Bricked]

shenbomo, 2023-09-17:
**Problem:**

- Cannot properly flash the firmware on the left module, all LEDs are off and it stays in an unresponsive state, with no Bluetooth LED flashing, and no key responses.

- Behavior change: After flashing the "Setting Reset" onto the left module, it would NOT immediately reboot into the bootloader mode anymore as before or as the right module would. Power-cycling by reconnecting the left module via USB-C, the 3 LEDs would flash white once quickly and then turn unresponsive, no key is registered, needless to say, the right module would not be able to connect with the left (3 LEDs flashing red).

- However, after flashing the "Setting Reset" onto the left module, it has to wait for around 15 - 20 minutes before the left module eventually reboots into the bootloader and is recognized by the PC  (3 solid green LEDs and USB drive mounted).

- After every power-cycling (disconnecting and reconnecting using USB), it has to wait for around 15 - 20 minutes before the left module eventually comes back online (recognized by the PC) with the profile LED flashing(ready to pair), and registering key, however, it immediately goes unresponsive when I turn on the right module, not able to sync with the right module.

**Things I have tried:** (NONE of them worked)

- Following precisely each step here: https://github.com/KinesisCorporation/Adv360-Pro-ZMK#overview
- Flashed Settings Reset v2.0/v3.0, then flashed the v2.0/v3.0 firmware respectively
- Flashed Settings Reset Erase Memory, then flashed the v2.0/v3.0 firmware respectively
- Flashed Settings Reset Erase Memory, then flashed Settings Reset v2.0/v.3.0, then flashed the v2.0/v3.0 firmware respectively
- Tried the firmware files from https://kinesis-ergo.com/support/kb360pro/#firmware-updates
- Tried the firmware files built by myself via GitHub Actions from https://github.com/KinesisCorporation/Adv360-Pro-ZMK
- Different USB cables
- Different machines (2 Windows 11 PCs, and 1 MacBook Pro M1)

**Speculation:**

- The hardware seems intact since I am still able to get into the bootloader mode consistently by double-clicking the reset button on both of the left and right modules. After 15-20 minutes of waiting after a power cycle, the left module registers keystrokes via a USB connection, however not being able to sync with the right module.

- Maybe the bootloader itself is messed up and requires a re-install or the SoftDevice s140 of the nRF52840 SoC (I am very new to ZMK and ZephyrOS, this is purely a baseless guess).

I am wondering if someone has run into similar issues before where flashing the Settings Reset UF2 and firmware UF2 messes up the left module like this.

I could not find any information related to fixing this kind of behavior anywhere, seems most problems can be fixed by flashing the reset and the firmware UF2 files.

Any help is much appreciated!! Thanks in advance!!

ReFil, 2023-09-18:
Hi, you could try and reinstall the bootloader but it sounds like a hardware issue to me.

Please open a ticket at: https://kinesis-ergo.com/support/kb360pro/#ticket for hardware related issues

shenbomo, 2023-09-18:
Thanks for the reply. I have already opened a ticket. In the meantime I am
curious to know if there is any documentation for reinstall the bootloader?

And also the keyboard worked fine right out of the box, only stopped
working after flashing the firmware for a few times, it was working fine
the first couple of times that I flashed the firmware. If your gut feeling
is correct, does that mean flashing the firmware might damage the hardware?
I was sure not disconnecting the USB during each flash.

Thanks again!

ReFil, 2023-09-19:
It is unlikely that flashing the keyboard will damage the hardware. The bootloader is very resilient to corrupt data transfers

shenbomo, 2023-09-19:
> It is unlikely that flashing the keyboard will damage the hardware. The bootloader is very resilient to corrupt data transfers

Thanks! That's what I assumed as well.
BTW in case you missed it, have you come across any publicly available tools, docs, and firmware files that can perform a lower-level bootloader or borad re-flash while forking and customizing the firmware for ADV360Pro by any chance? 
I am eager to try it out myself.
Much apprecietd!!

KinesisCorporation, 2023-09-28:
@shenbomo Can you update us on the status of your keyboard please? Hopefully no news is good news.

shenbomo, 2023-10-14:
I have contacted tech support, and they are helping me with the RMA process.
Thanks for following up!

ReFil, 2023-10-14:
No worries, thank you for replying. If you're going through the RMA process I will close this issue out now

closed 2023-10-14 by ReFil