KinesisCorporation/Adv360-Pro-ZMK #230: Bluetooth Connectivity Fails At macOS Login Screen

patricksalo, 2023-09-20:
**Prerequisites**
macOS 13.5.2 Ventura (or IIRC 13.5.1 as well)
Commit from Sep 7, 2023

**Description of Problem**
Starting with version 3.0 of the firmware, the Adv360-Pro no longer connects to my MacBook Pro (M1) at login upon cold boot or restart. Note that this is the login screen _prior_ to macOS loading. My Logitech MX Ergo trackball has always worked and confirms Bluetooth is active and working at the hardware and pre-OS boot level.

Today I had time to revert to my prior 2.x firmware and as soon as I did so this issue went away. So there's definitely something funky with the 3.x firmware and Bluetooth.

**Additional Comments Semi-Unrelated**
As an aside, when I originally updated to the 3.x firmware, I spent a really long time, standing on my head, running around in circles, etc. before I could get the keyboard to reconnect to macOS and even when doing so I had to struggle through trying to get the right half to connect properly to the left. I probably spent 30-60 mins before things magically worked.

KinesisCorporation, 2023-09-20:
We are seeing this problem with 13.5.1 as well. We will investigate.

Sorry for the trouble and thanks for the incredibly clear documentation!

ReFil, 2023-09-21:
Is this the filevault/ full disk encryption screen? 

try deleting this line https://github.com/KinesisCorporation/Adv360-Pro-ZMK/blob/45fca6f6010b50cec3e7f8df4fe4af4bad1e470e/config/boards/arm/adv360/adv360_left_defconfig#L52 
and see if it still presents an issue

patricksalo, 2023-09-21:
Great suggestion @ReFil and it looks like that solved my problem. Yes, this is the FileVault login screen at EFI prior to booting macOS. I suspect this also solved all my other woes too in that...

1. Both halves had zero issues connecting to each other
2. Upon rebooting my Mac, the keyboard once again connected and could be used at the pre-OS boot login

Additionally this solves another issue I had seen and the experience was such a night and day difference that I'll expand here. I'm using my Adv360-Pro also with a Windows 11 system. Under the prior 3.0 firmware, getting it to pair was quite a pain as well. Not nearly as bad as the Mac at 30-60 mins but maybe 15-30 mins of various reboots of the system, keyboard, multiple pairing tries, etc. After making this change and pairing, Bluetooth instantly connected on the first try.

ReFil, 2023-09-22:
I suspect an update changed where the bluetooth LE privacy keys were stored, i suppose it makes sense for them to only be decrypted after passing the efi encryption screen. One possible side effect of disabling privacy on the keyboard is that the keyboard might periodically fail to reconnect to the mac as the mac has cycled to a new private address. If your issue is solved now i'll close it out

closed 2023-09-22 by ReFil

ReFil, 2023-10-02:
@patricksalo Just following up on this, have you had any connectivity issues requiring re-pairing to the mac since disabling BT privacy?

patricksalo, 2023-10-02:
So far so good and everything has been extremely solid for both Mac and PC.