KinesisCorporation/Adv360-Pro-ZMK #238: Can't connect to keyboard with V3.0 firmware on linux

zbw8388, 2023-10-01:
Hi, I have followed the instruction to update my keyboard to V3.0 yet my laptop now have trouble connecting to the keyboard. I'm running Fedora Linux 38 with Linux 6.5.5-200.fc38.x86_64. 

So far, I've tried 
1. removing `CONFIG_BT_PRIVACY=y` https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/198#issuecomment-1655661005 
2. using beta firmware https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/198#issuecomment-1691427952
3. combining the two above
4. using Flash Erase (https://kinesis-ergo.com/support/kb360pro/#firmware-updates) and then repeat 3

and none of them worked. I tried going back to V2.0 and it seems to be working fine.

I turned on usb logging for the left module with V3.0 firmware (https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/104#issuecomment-1613496432), and here are the logs: 

When just upgraded from V2.0 with a (maybe?) working bluetooth security key (https://zmk.dev/docs/behaviors/bluetooth):
<details><p>

```
[00:22:29.923,431] <dbg> zmk: backlight_event_listener: event called
[00:22:29.923,553] <dbg> zmk: disconnected: Disconnected from <mac> (public) (reason 0x05)
[00:22:29.923,583] <dbg> zmk: update_advertising: advertising from 2 to 2
[00:22:30.206,237] <dbg> zmk: split_central_connected: SKIPPING FOR ROLE 1
[00:22:30.206,268] <dbg> zmk: connected: Connected thread: 0x200052d8
[00:22:30.206,359] <dbg> zmk: connected: Connected <mac> (public)
[00:22:30.207,733] <dbg> zmk: update_advertising: advertising from 0 to 2
[00:22:30.356,811] <err> zmk: Security failed: <mac> (public) level 1 err 2
[00:22:30.463,439] <dbg> zmk: split_central_disconnected: Disconnected: <mac> (public) (reason 5)
[00:22:30.463,470] <dbg> zmk: rgb_underglow_event_listener: event called
[00:22:30.463,500] <dbg> zmk: backlight_event_listener: event called
[00:22:30.463,592] <dbg> zmk: disconnected: Disconnected from <mac> (public) (reason 0x05)
[00:22:30.463,623] <dbg> zmk: update_advertising: advertising from 2 to 2
[00:22:30.646,575] <dbg> zmk: split_central_connected: SKIPPING FOR ROLE 1
[00:22:30.646,575] <dbg> zmk: connected: Connected thread: 0x200052d8
[00:22:30.646,667] <dbg> zmk: connected: Connected <mac> (public)
[00:22:30.647,979] <dbg> zmk: update_advertising: advertising from 0 to 2
[00:22:30.806,854] <err> zmk: Security failed: <mac> (public) level 1 err 2
[00:22:30.912,902] <dbg> zmk: split_central_disconnected: Disconnected: <mac> (public) (reason 5)
[00:22:30.912,933] <dbg> zmk: rgb_underglow_event_listener: event called
[00:22:30.912,933] <dbg> zmk: backlight_event_listener: event called
[00:22:30.913,055] <dbg> zmk: disconnected: Disconnected from <mac> (public) (reason 0x05)
[00:22:30.913,085] <dbg> zmk: update_advertising: advertising from 2 to 2
[00:22:31.275,085] <dbg> zmk: split_central_connected: SKIPPING FOR ROLE 1
[00:22:31.275,115] <dbg> zmk: connected: Connected thread: 0x200052d8
[00:22:31.275,207] <dbg> zmk: connected: Connected <mac> (public)
[00:22:31.276,397] <dbg> zmk: update_advertising: advertising from 0 to 2
[00:22:31.436,828] <err> zmk: Security failed: <mac> (public) level 1 err 2
[00:22:31.588,470] <dbg> zmk: split_central_disconnected: Disconnected: <mac> (public) (reason 5)
[00:22:31.588,531] <dbg> zmk: rgb_underglow_event_listener: event called
[00:22:31.588,531] <dbg> zmk: backlight_event_listener: event called
[00:22:31.588,653] <dbg> zmk: disconnected: Disconnected from <mac> (public) (reason 0x05)
[00:22:31.588,684] <dbg> zmk: update_advertising: advertising from 2 to 2
```

</p></details>
After removing devices from laptop and keyboard to start afresh:

<details><p>

```
[00:25:30.145,355] <dbg> zmk: split_central_connected: SKIPPING FOR ROLE 1
[00:25:30.145,385] <dbg> zmk: connected: Connected thread: 0x200052d8
[00:25:30.145,477] <dbg> zmk: connected: Connected <mac> (public)
[00:25:30.146,667] <dbg> zmk: update_advertising: advertising from 0 to 2
[00:25:30.229,003] <dbg> zmk: auth_pairing_accept: role 1, open? yes
[00:25:30.510,467] <err> zmk: Security failed: <mac> (public) level 1 err 9
[00:25:30.523,498] <dbg> zmk: split_central_disconnected: Disconnected: <mac> (public) (reason 42)
[00:25:30.523,529] <dbg> zmk: rgb_underglow_event_listener: event called
[00:25:30.523,559] <dbg> zmk: backlight_event_listener: event called
[00:25:30.523,651] <dbg> zmk: disconnected: Disconnected from <mac> (public) (reason 0x2a)
[00:25:30.523,681] <dbg> zmk: update_advertising: advertising from 2 to 2
```

</p></details>

Is there a way to fix it? Any help is appreciated!

ReFil, 2023-10-01:
Can try removing these lines too.

https://github.com/KinesisCorporation/Adv360-Pro-ZMK/blob/V3.0/config/boards/arm/adv360/adv360_left_defconfig#L53-L57

I've had no issues connecting to my Linux laptop using elementary OS which is based off Ubuntu 20.04. Can you collect dmesg logs when you try and pair?

zbw8388, 2023-10-01:
Thanks for the help! I tried deleting those lines but it still didn't work.

This issue doesn't seem to appear in an old kernel. Guess there is some regression going on. I'll just use that version for a workaround.

ReFil, 2023-10-01:
That's odd, thanks for the information, I'll see if there's anything that can be done to solve the issue

ReFil, 2023-10-10:
I don't think it's a 360 specific issue, I've seen a report of Linux connectivity issues after updating in the main ZMK discord by another user, happy to teopn

ReFil, 2023-10-10:
According to this user: https://discord.com/channels/719497620560543766/1158881550788734976/1161346978484060221

It was an update to the Bluetooth adapter drivers in the Linux-firmware package that broke things and downgrading that package rectified the issue

ReFil, 2023-10-13:
Sounds good, I can add that to the documentation somewhere

closed 2023-10-13 by ReFil