KinesisCorporation/Adv360-Pro-ZMK #239: Entire column of keys not working Fn, Shift, Esc, Tab, Plus/Equals.

rustyconover, 2023-10-01:
Hello, I've loved my Advantage 360 ZMK keyboard so far, but today an entire column of keys stopped working. 

I use firmware version 3.0, but my: Fn, Shift, Esc, Tab, Plus/Equals (can't type it) no longer respond to keypresses.  I'm using Ventura 13.5.2.

Even when I press the fn key the LED no longer changes color.

All over keys work perfectly.

Please advise if there is some software change to be made or this has been seen before.

ReFil, 2023-10-01:
This sounds like a hardware issue. Please open a ticket with Kinesis support on the website. https://kinesis-ergo.com/support/kb360pro/#ticket

ReFil, 2023-10-13:
Just following up on this, have you had this issue sorted?

KinesisCorporation, 2023-10-20:
@rustyconover We have received your ticket and will handle your RMA directly. Sorry for the inconvenience.

closed 2023-10-20 by KinesisCorporation