KinesisCorporation/Adv360-Pro-ZMK #243: Build failure with keymaps contains LS(LC(LA(LGUI))) / LS(LC(LA))

PongPong, 2023-10-07:
Hi , i 've tried to update Hyper [ LS(LC(LA(LGUI))) ] and Meh [ LS(LC(LA)) ] keys via GUI but failed.

https://kinesiscorporation.github.io/Adv360-Pro-GUI/

```bash
Run west build -s zmk/app -d build/left -b adv360_left -- -DZMK_CONFIG="${GITHUB_WORKSPACE}/config"
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/zmk/app
-- Adding ZMK config directory as board root: /__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/config
-- ZMK Config directory: /__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/config
-- Using keymap file: /__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/config/adv360_left.keymap
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.10") found components: Interpreter 
-- Cache files will be written to: /__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/zephyr/.cache
-- Zephyr version: 3.2.0 (/__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/zephyr)
-- Found west (found suitable version "0.1[4](https://github.com/PongPong/Adv360-Pro-ZMK/actions/runs/6437862020/job/17483628667#step:8:5).0", minimum required is "0.7.1")
-- Board: adv360_left
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.1[5](https://github.com/PongPong/Adv360-Pro-ZMK/actions/runs/6437862020/job/17483628667#step:8:6).2 (/opt/zephyr-sdk-0.15.2)
-- Found toolchain: zephyr 0.15.2 (/opt/zephyr-sdk-0.15.2)
-- Found Dtc: /opt/zephyr-sdk-0.15.2/sysroots/x8[6](https://github.com/PongPong/Adv360-Pro-ZMK/actions/runs/6437862020/job/17483628667#step:8:7)_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/config/boards/arm/adv360/adv360_left.dts
-- Found devicetree overlay: /__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/config/adv360_left.keymap
devicetree error: /__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/config/adv360.keymap:44 (column 445): parse error: expected number or parenthesized expression
-- In: /__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/build/left/zephyr, command: /usr/bin/python3.[8](https://github.com/PongPong/Adv360-Pro-ZMK/actions/runs/6437862020/job/17483628667#step:8:9);/__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/zephyr/scripts/dts/gen_defines.py;--dts;/__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/build/left/zephyr/zephyr.dts.pre;--dtc-flags;'';--bindings-dirs;/__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/zmk/app/drivers/zephyr/dts/bindings;/__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/zmk/app/dts/bindings;/__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/zephyr/dts/bindings;--header-out;/__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/build/left/zephyr/include/generated/devicetree_generated.h.new;--dts-out;/__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/build/left/zephyr/zephyr.dts.new;--edt-pickle-out;/__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/build/left/zephyr/edt.pickle;--vendor-prefixes;/__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/zmk/app/dts/bindings/vendor-prefixes.txt;--vendor-prefixes;/__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/zephyr/dts/bindings/vendor-prefixes.txt
CMake Error at /__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/zephyr/cmake/modules/dts.cmake:230 (message):
  gen_defines.py failed with return code: 1
Call Stack (most recent call first):
  /__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/zephyr/cmake/modules/zephyr_default.cmake:108 (include)
  /__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:[9](https://github.com/PongPong/Adv360-Pro-ZMK/actions/runs/6437862020/job/17483628667#step:8:10)2 (include_boilerplate)
  CMakeLists.txt:[15](https://github.com/PongPong/Adv360-Pro-ZMK/actions/runs/6437862020/job/17483628667#step:8:16) (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/build/left -GNinja -DBOARD=adv360_left -DZMK_CONFIG=/__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/config -S/__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/zmk/app
Error: Process completed with exit code 1.
```

ReFil, 2023-10-07:
LA is a modifier code not a key code. If you want a key that does left shift + lctrl+ lalt it needs to be configured as LS(LC(LALT)).

PongPong, 2023-10-07:
@ReFil , I selected those two mappings from GUI, when I type the words starts with LS , it auto complete the input field, just wonder if it is a typo.

ReFil, 2023-10-07:
It might be expecting you to put another binding in, either way if you correct it to LALT it should work

ReFil, 2023-10-07:
Just following up, did that solve the problem?

PongPong, 2023-10-07:
yes , it builds successfully, now I will try to flash the firmware to see if it works , thanks and appreciate your help.

PongPong, 2023-10-08:
@ReFil it works perfectly after flashing the firmware , thanks :)

closed 2023-10-08 by ReFil