KinesisCorporation/Adv360-Pro-ZMK #249: No input with latest ZMK update

DDRBoxman, 2023-10-22:
I was trying to build a firmware with the update from yesterday but on macOS I wasn't getting any input till I reverted the change.
[](https://github.com/DDRBoxman/Adv360-Pro-ZMK/commits/V3.0)

ReFil, 2023-10-22:
Was this over USB, BLE or both? With a minor update like that you might need to unpair and repair to the host

DDRBoxman, 2023-10-22:
I ran into this with both USB and BLE, I also went through and cleared out the connection in the Bluetooth and flashed the setting reset firmware between updates.

ReFil, 2023-10-22:
I can't replicate this issue on my test computers, windows or an M1 MacBook Pro running MacOS Sonoma. Have you tried connecting it to a phone or other computer?

mychmcg, 2023-10-22:
The only input I can get is from using the Adv360 Pro factory default firmware (v3.0 10-23-23). Otherwise both my custom keymap(created for the first time last night) and a pure clone of this repo give this error after flashing, on windows 11.
[image]

 Bluetooth works fine with factory defaults as well. With the bad firmware I can tell my layers activate properly and can even clear bluetooth. But my pc cannot hear any of it

ReFil, 2023-10-22:
Hi all, many thanks for the reports. A commit that may be causing the issue has been identified and rolled back. Please try rebuilding your personal repos now and reflashing the boards. A settings-reset shouldn't be required but you will need to unpair, BT_CLR and repair to the computer

mychmcg, 2023-10-22:
Thank you for the fix!

ReFil, 2023-10-22:
Has that solved your issue?

mychmcg, 2023-10-22:
Yes, my custom keymap is inputting to my PC correctly

DDRBoxman, 2023-10-23:
Cool, seems like that fixed it, thanks!

closed 2023-10-23 by ReFil