KinesisCorporation/Adv360-Pro-ZMK #251: `CONFIG_BT_PRIVACY` cannot be set to `y`

ickc, 2023-10-24:
C.f. #248. This behavior is documented. The issue is opened only to track when this is resolved.

ReFil, 2023-10-25:
I dont think this should be tracked, further experimentation with BT privacy has indicated that on the latest MacOS updates disabling it is the way to go, this also improves the behaviour if using filevault. If Privacy gets re-enabled in the future that will be documented but at the moment there are no plans to reinstate privacy support

closed 2023-10-25 by ReFil

ickc, 2023-10-25:
Thanks for the feedback. Could you add this in the doc too? It wasn't clear immediately what's the best practice here and I thought I was a regression but sounds like it's actually better in some cases.

I'd also appreciate a few words on `CONFIG_BT_BAS`, as it is default to n now. Some background on potential problem with that enabled will be helpful.

ReFil, 2023-10-27:
That sounds like a good idea, i'll work on that

ReFil, 2023-11-02:
This is being addressed in #260