KinesisCorporation/Adv360-Pro-ZMK #256: Upgrade instructions from V2.0 to V3.0 is not clear.

eugene8080, 2023-10-30:
Suggestion to revise the v2.0 to v3.0 firmware upgrade.
The current issue with the instructions is that it requires combining instructions from three sources:
https://kinesis-ergo.com/support/kb360pro/#firmware-updates
https://kinesis-ergo.com/wp-content/uploads/Step-by-Step-Advantage360-Professional-Firmware-Installation-Instructions-KB360-PRO_v11-10-22.pdf
https://github.com/KinesisCorporation/Adv360-Pro-ZMK/blob/V3.0/README.md
Some of the issues with this include: 1) It is not clear in the different documents that both settings-reset-v2.0.uf2 and settings-reset-v3.0.uf2 need to be flashed. 2) The modules do not disconnect per the instructions after the settings-reset files, and ejecting the drives have no effect. The user is forced to turn off the modules or to keep loading settings-reset-v2.0.uf2, settings-reset-v3.0.uf2, then left.uf2/right.uf2 before they disconnect. 

Suggested Instruction, but I'm not sure that this is correct.
1) Extract the firmwares from the archive downloaded from the GitHub build job (If using the cloud builder) or the firmware folder (If building locally). Download settings-reset-v2.0 and settings-reset-v3.0
2) Connect the left side keyboard to USB.
3) Option 1: Press Mod+macro1 to put the left side into bootloader mode; it should attach to your computer as a USB drive. Or Option 2: place the left module into bootloader mode by using a paperclip to quickly DOUBLE-CLICK the Reset Button. The Indicator LEDs with flash white and then change to solid green to indicate bootloader mode.
4) Copy settings-reset-v2.0.uf2 to the USB drive. The backlight will ligth and the 3 LEDs will be a solid green.
5) Copy settings-reset-v3.0.uf2 to the USB drive. The backlight will ligth and the 3 LEDs will be a solid green.
6) Copy left.uf2 to the USB drive and it will disconnect.
7) Power off both keyboards (by unplugging them and making sure the switches are off).
8) Connect the right side keyboard to USB.
9) Place the right module into bootloader mode by using a paperclip to quickly DOUBLE-CLICK the Reset Button. The Indicator LEDs with flash white and then change to solid green to indicate bootloader mode.
10) Copy settings-reset-v2.0.uf2 to the USB drive. The backlight will ligth and the 3 LEDs will be a solid green.
11) Copy settings-reset-v3.0.uf2 to the USB drive. The backlight will ligth and the 3 LEDs will be a solid green.
12) Copy right.uf2 to the USB drive and it will disconnect.
13) Power off the right keyboard (by unplugging it and making sure the switche is off).
14) Turn on both modules and reconnect to computer's bluetooth. If you previously connected the Keyboard, you will  need to remove the old bluebooth connection and add it again.

KinesisCorporation, 2023-10-30:
It's a good point and something we're looking at right now. The main issue is there is a the quick and easy way to flash an update that works 95% of the time, and then there is the "something didn't quite work right" way that is much longer and should work 99.9% of the time (absent a hardware issue).

Power users will want to use the quick and easy way most of the time since they will be more adept at troubleshooting if the update fails. But everyone else will probably be better served "playing it safe" each time with the full process even though it may take an extra ~2 minutes.

eugene8080, 2023-11-01:
Noted. Might I suggest an alternative then? How about you set settings-reset-v2.0.uf2 and settings-reset-v3.0.uf2 to eject the USB drive after flashing? This was the behavior in settings-reset-v1.0.uf2. This way, at least the behavior matches your instructions.

ReFil, 2023-11-01:
They do eject the drive and reboot the board, it is an artifact of the operating system that it treats it like it is not ejected, we will see what can be done to improve this behaviour

KinesisCorporation, 2023-11-01:
> FWIW: Glove80 has this same issue with MacOS. When I flash that keyboard, it ejects but MacOS thinks it isn't and gives me the same message the 360 is victim to.

Same with Kinesis SmartSet devices. It's just how macOS deals with these kind of drives:(

closed 2023-11-06 by eugene8080