KinesisCorporation/Adv360-Pro-ZMK #257: reset firmware update

hyojii00, 2023-10-31:
Contrary to the guide document, since updating from the right side first, the LED on the right device is flashing red quickly and does not work.
I can't connect to my macbook through finder to clear the update file because it doesn't connect separately. Does anyone know a solution?

ReFil, 2023-10-31:
https://github.com/KinesisCorporation/Adv360-Pro-ZMK#overview There are backup hardware reset buttons you can press with a paperclip documented in the user manual

closed 2023-10-31 by ReFil

KinesisCorporation, 2023-10-31:
[image]

hyojii00, 2023-10-31:
@ReFil When I press both reset buttons, all the LEDs blink once and then flash red quickly again.
My guess is that the right device is not finding the left device.
Is there any way to revert it via the reset button?

KinesisCorporation, 2023-10-31:
You only need to press one reset button, but that key module has to be connected to the PC over USB. Reset and Update the left module, then Reset and update the right module.

ReFil, 2023-10-31:
Please note the reset button must be pressed twice in rapid succession to enter the usb bootloader

hyojii00, 2023-10-31:
the right module I can reset. but the left side, after click reset button, leds is not blink and turn green. there are just white.

KinesisCorporation, 2023-10-31:
What number does your serial number start with 1, 2, 3 etc?

Please try using a different USB cable to connect the left module to PC. If it doesn't work the first time then rotate the cable it 180 degrees and try again. Its possible the USB connect is bad. If it still doesn't work please submit a trouble ticket directly to Kinesis and include a video and your serial number. Sorry!

hyojii00, 2023-11-01:
감사합니다. 겨우 두 기기간의 페어는 연결했습니다. 더이상 빨간색 led 는 깜빡이지 않습니다.
그러나 맥북과의 블루투스 연결이 되지 않는데 이번에도 도움을 요청해도 될까요?
블루투스 목록에서는 adv 가 확인되지만 connect 가 되질 않습니다.
업무끝나고 다시 시도해보겠습니다.

KinesisCorporation, 2023-11-01:
Press the Mod + Windows key to erase any possible pairing in the active Profile on the keyboard. That should ensure the the Profile LED on the left side is flashing "Fast" and the keyboard is ready to pair that Profile. Then you just need to find the "Adv360 Pro" device in your Macbook BT menu and select Connect. If you previously paired the keyboard with your Macbook we recommended "Forget this Device" and then pair from scratch.

Please submit a trouble ticket directly to Kinesis if you need additional assistance. 

https://kinesis-ergo.com/support/contact-a-technician/