KinesisCorporation/Adv360-Pro-ZMK #258: Indicator lights always solid on right half

davidsgrogan, 2023-11-01:
The right indicator lights are always on when the backlight is on. Even powering off left half doesn't make the right indicator lights flash red.

[image]

This started a few days ago when I was trying to check the battery levels. I was on firmware V2. Today I installed V3, still had the problem, so installed `settings-reset-v3.0.uf2` to both halves. The right indicator lights came on and stayed on even before installing V3 back. I've since installed V3 and the lights remain on.

Mod+③+④ makes the right half's backlight flash, but the LEDs come right back on.[^1]

How do I get those LEDs to work again?

[^1]: That suggestion is from https://www.reddit.com/r/kinesisadvantage/comments/xl4adf/360_pro_stuck_with_3_green_leds_solid/ which hints that there used to be an entry for this problem on the troubleshooting page?

ReFil, 2023-11-01:
Hi, it seems like the indicator lights have gotten stuck in the battery indication mode. A fix for this behaviour has just been incorporated into the beta branch, would you be able to test that and provide feedback please?

ReFil, 2023-11-01:
To test with the beta branch you can follow the instructions here: https://github.com/KinesisCorporation/Adv360-Pro-ZMK#beta-testing

KinesisCorporation, 2023-11-01:
> To test with the beta branch you can follow the instructions here: https://github.com/KinesisCorporation/Adv360-Pro-ZMK#beta-testing

And same with Kinesis SmartSet devices. Unfortunately its just how macOS deals with these kind of drives.

ReFil, 2023-11-04:
Hi, just following up on this, have you been able to get it back to normal?

ReFil, 2023-11-06:
Hi, just following up again, if we don't hear back from you in 48 hours we'll close this issue out as sooved

davidsgrogan, 2023-11-07:
Yes, beta worked, thanks. How will I know when the fix has been included in the stable branch and I should remove `-beta` from west.yml?

ReFil, 2023-11-07:
I will update this  issue when it's done, it should happen within the next few days, there's onlyy one outstanding change that must be made before it's included

closed 2023-11-07 by ReFil

ReFil, 2023-11-08:
The changes are integrated into the stable branch, you can revert to adv360-z3.2-2