KinesisCorporation/Adv360-Pro-ZMK #259: Repeating Keys issue. Firmware: V3.0 OS: Windows 11

eugene8080, 2023-11-02:
Separating this issue from the [bluetooth regression thread](https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/198) as suggested.

After upgrading from v2.0 to v3.0 firmware. I keep getting a repeat keys issue. It is frequent enough that the keyboard is not usable. I thought the repeating keys issue went away after using adv360-z3.2-beta, but it came back again.
When I flashed the [v2.0 firmware](https://kinesis-ergo.com/wp-content/uploads/Adv360-Pro-Factory-Default-Firmware-v2.0-11-1-22.zip), I have no repeating keys issue at all. Therefore, I do no suspect it is a hardware issue.
Where can I find the old v2.0 files on Github so I can make changes to the keymap and use 2.0 until 3.0 becomes usable on my computer?

===============================================
Everything I have tried for v3.0 firmware has resulted in repeating keys.

I haved tried:
turning off my Logitech mouse (@yanshay)
disconnecting the logitech wifi dongle
turning off bluetooh on my phone and computer before turning on the adv360

Drivers:
All bluetooth and windows drivers are up to date.

Settings reset, i have used:
settings-reset.uf2 released 2023-10-30
settings-reset-v1.0.uf2, settings-reset-v2.0.uf2, settings-reset-v3.0.uf2 in sequence

Firmware I have now tested:
Adv360 Pro Factory Default Branch 3.0 (v10-23-23)
Adv360 Pro Factory Default Branch 3.0 (v7-6-23)
adv360-z3.2-2
adv360-z3.2-beta

===========================================
Keyboard: Kinesis Advantage360 Pro
Computer: Surface Laptop Studio, 11th Gen Intel(R) Core(TM) i7-11370H @ 3.30GHz
Bluetooth: Intel Wifi-6 AX200 160MHz
OS: Windows 11 Home 22H2

KinesisCorporation, 2023-11-02:
Hi Eugene,

Thanks for creating a new issue and all the work that went into documenting what you are experiencing. We're really sorry this is still happening. @ReFil has double checked and the fix didn't get lost in the wires during the 3.0 update and they have not been able to replicate it so we're not exactly sure what's going on.

Here is a link to the 2.0 Factory Default firmware from June. Would you mind installing this and using it for a period long enough to determine whether the repeating keys is happening? If it is, we'll re-release the 2.0 branch for you and continue to investigate.

https://kinesis-ergo.com/download/advantage360-professional-factory-default-6-1-23/

ReFil, 2023-11-02:
Hi Eugene, Would you be able to check which version of the Bluetooth drivers you're using? Afaik Intel Bluetooth version 22.250 is known to be troublesome with other ZMK powered boards

eugene8080, 2023-11-03:
> Hi Eugene, Would you be able to check which version of the Bluetooth drivers you're using? Afaik Intel Bluetooth version 22.250 is known to be troublesome with other ZMK powered boards

Intel Driver Version: 22.250.1.2
If the updated Intel Driver is the issue, that might explain why adv360-z3.2-beta worked for some time then stopped working again.

eugene8080, 2023-11-03:
> Hi Eugene,
>
> Thanks for creating a new issue and all the work that went into documenting what you are experiencing. We're really sorry this is still happening. @ReFil has double checked and the fix [...]

Yes this works even with Intel driver 22.250.1.2

eugene8080, 2023-11-03:
Passed:
Intel 22.250.1.2 with Adv360-Pro-Factory-Default-Firmware-v2.0-11-01-22
Intel 22.250.1.2 with Adv360-Pro-Factory-Default-Firmware-v2.0-06-01-23
Intel 22.240.0.2 with adv360-z3.2-beta
Intel 22.240.0.2 with Adv360 Pro Firmware (Factory Default 10-23-23)
Intel 22.240.0.2 with adv360-z3.2-2
Android 13 with Adv360 Pro Firmware (Factory Default 10-23-23)
Android 13 with adv360-z3.2-2

Failed:
Intel 22.250.1.2 with Adv360-Pro-Factory-Default-Firmware-v3.0-10-23-23
Intel 22.250.1.2 with Adv360-Pro-Factory-Default-Firmware-v3.0-07-06-23
Intel 22.250.1.2 with adv360-z3.2-beta (10-28-23)
Intel 22.250.1.2 with adv360-z3.2-2
Intel 22.250.1.2 with adv360-z3.2-beta
Intel 22.230.0.8 with adv360-z3.2-beta
Android 13 with adv360-z3.2-beta

ReFil, 2023-11-03:
Thank you very much for the intensive testing, is much appreciated. It seems like there's a combination of factors at play here, what could help is changing the ble connection parameters, could you try removing these 5 lines from your config file and trying again?

https://github.com/KinesisCorporation/Adv360-Pro-ZMK/blob/V3.0/config/boards/arm/adv360/adv360_left_defconfig#L53-L57

eugene8080, 2023-11-03:
Do I remove the lines by putting a "#" in front of each line?

Should I test all of them, or just the newer firmware that failed? I'm assuming you're most interested in the newest Intel drivers with the newest adv360 firmware right?

ReFil, 2023-11-03:
A # in front will comment them out, yes

Please just test the newest firmware and Intel driver thanks, either adv360-z3.2-beta or adv360-z3.2-2

eugene8080, 2023-11-05:
> Hi Eugene, Would you be able to check which version of the Bluetooth drivers you're using? Afaik Intel Bluetooth version 22.250 is known to be troublesome with other ZMK powered boards

Haven't tried your suggestion yet, but saw this over on the ZMK issues 

https://github.com/zmkfirmware/zmk/issues/805#issuecomment-1792680673

eugene8080, 2023-11-06:
Here is the new testing:

Passed:
Android 13 with adv360-z3.2-2 adv360_left_defconfig#L53-L57
Intel 22.240.1.2 with adv360-z3.2-2 adv360_left_defconfig#L53-L57
Intel 22.240.1.2 with adv360-z3.2 beta adv360_left_defconfig#L53-L57
Intel 22.250.1.2 with adv360-z3.2-2 adv360_left_defconfig#L53-L57
Intel 22.250.1.2 with adv360-z3.2-beta adv360_left_defconfig#L53-L57

Failed:
Android 13 with adv360-z3.2 beta adv360_left_defconfig#L53-L57

This is great news, I'll start to use adv360-z3.2-2 adv360_left_defconfig#L53-L57.
@ReFil please note there is always an issue with Android on the adv360-z3.2-beta firmware. There is no keystroke response at all.

ReFil, 2023-11-06:
Thanks a lot for the testing, if no issues present in 24h or so it would be much appreciated if you can close this issue out

Thanks for the feedback on android. I suspect this is a consequence of a potential fix to NKRO with extra keys, I'll make that option configurable

eugene8080, 2023-11-06:
Fixed the issue with deleted these lines:
https://github.com/KinesisCorporation/Adv360-Pro-ZMK/blob/V3.0/config/boards/arm/adv360/adv360_left_defconfig#L53-L57

eugene8080, 2023-11-07:
Sorry, to have to reopen this. I was using:
Intel 22.250.1.2 with adv360-z3.2-2 adv360_left_defconfig#L53-L57
The repeating keys issue is much improved but I experience it after several hours of use. It is MUCH better than before when it was barely usable, but the problem persists. 
I will switch back to Intel 22.240.1.2 for now.

ReFil, 2023-11-07:
Hi Eugene, ok sounds good, it might be worth changing the title of this issue to explicitly mention the affected intel driver? I'm tracking some pending changes in upstream ZMK that should improve the BLE stack and may fix this issue

eugene8080, 2023-11-09:
> Hi Eugene, ok sounds good, it might be worth changing the title of this issue to explicitly mention the affected intel driver? I'm tracking some pending changes in upstream ZMK that should improve the BLE stack and may fix this issue

Changed

FlippingBinary, 2023-12-03:
I spoke too soon. Just experienced a brief disconnect. Luckily I didn't have a key pressed when it happened.

FlippingBinary, 2023-12-04:
It's definitely not worse. It might be better. Sorry, it's kind of hard to say with an intermittent problem like this.

ReFil, 2023-12-04:
I totally understand, thanks for being honest about it. Intermittent issues like this are really challenging to replicate on my end too because I've only got so many computers I can test with. When it disconnects is it just that it won't respond for a bit then automatically reconnects and let's you keep going? Or do you need to go into the settings menu and reconnect manually/re-pair?

ReFil, 2023-12-08:
Thanks for the update, a momentary disconnect could happen for a few reasons but its good that it's not misbehaving right now (fingers crossed it stays that way)

FlippingBinary, 2023-12-12:
It happened again just now. The Bluetooth LED was blinking while a key was stuck repeating. I pulled the recent commits to the main repository and pushed one change onto mine to restore the `CONFIG_BT_CTRL_PHY_2M=n` setting that seemed to help. It looks like some other configuration changes made it into main that I was already using, but I didn't have `CONFIG_BT_PRIVACY=n` commented out, so that will be different this time around.

One week between events isn't as bad as it was, but it's still more than I'd like.

ReFil, 2024-01-07:
We've worked on a new beta firmware to improve connection reliability, this might help the momentary disconnect issue

If people affected by this issue could trial this firmware for a bit and provide feedback it would be much appreciated: [360pro-beta-1-6-24.zip](https://github.com/KinesisCorporation/Adv360-Pro-ZMK/files/13854902/360pro-beta-1-6-24.zip)

saito-sv, 2024-01-10:
> We've worked on a new beta firmware to improve connection reliability, this might help the momentary disconnect issue
>
> If people affected by this issue could trial this firmware for a bit and [...]

am going to give this a go, I am unable to connect to on my linux(fedora 39) machine.

FlippingBinary, 2024-01-10:
I don't think I've had a disconnect since my last comment. My firmware is only five commits behind main and the changes appear to be related to a version macro, so I don't think the beta firmware would introduce any related changes. Still, I went ahead and flashed the beta and had no trouble connecting to my Windows 10 Pro computer or my Manjaro Linux laptop. I report back if there are any problems.

ReFil, 2024-01-11:
The beta firmware has a lot of new differences, especially with regards to BLE improvements. it references a new base ZMK branch and a different config repo so the changes wouldn't show up on the main config

ReFil, 2024-01-14:
These configuration changes will be made upstream with the merging of #326 under the config option of `CONFIG_ZMK_BLE_EXPERIMENTAL_FEATURES`. It might cause issues if users have already made the changes separately and thus when it's merged it would be a good idea to remove the `CONFIG_BT_CTLR_PHY_2M=n` and `CONFIG_BT_LL_SW_LLCP=y` lines where they occur

ReFil, 2024-01-14:
This PR has now been merged so this issue is being closed as solved, any feedback on the changes would be much appreciated

closed 2024-01-14 by ReFil