KinesisCorporation/Adv360-Pro-ZMK #261: cannot flash uf2

42x00, 2023-11-03:
Hi,

Everytime I copy uf2 file to the keyboard, the following error occurs:

```
The Finder can’t complete the operation because some data in “left.uf2” can’t be read or written.
(Error code -36)
```

I have tried:

- different mac os
- chmod 777 uf2 file
- different way to enter bootloader
- different uf2 files: Settings Reset, Factory Default Firmware

ReFil, 2023-11-03:
Hi, just to clarify is the board not updating at all? Error messages with finder do not necessarily mean the operation failed

closed 2023-11-04 by 42x00

IamGianluca, 2023-11-15:
I can confirm that often error messages in Finder (Mac OS) and Files (Linux) do not mean that the firmware update operation failed. After copying the .uf2 files on both the left and right sides, I saw this error, but the operation was successful.

[image]