KinesisCorporation/Adv360-Pro-ZMK #262: Bluetooth Connectivity Extremely Flaky Under macOS 14.1.0

patricksalo, 2023-11-05:
**Prerequisites**
macOS 14.1.0 Sonoma
Commit from Nov 2, 2023

**Description of Problem**
Since upgrading to macOS 14.1.0, my Kinesis 360 Pro has been extremely flaky with Bluetooth. I was previously on the firmware build from Sep 7, 2023 and after the privacy tweak #230 it has been extremely solid under 14.0.0. Currently the flakiness falls under two semi-repeatable scenarios:

1. Password Entry After Display Sleep: If I return to my system and move my trackball to get to the macOS unlock screen, the keyboard shows connected but all keystrokes aren't sent. It took me a bit to figure this out, but these keystrokes are sent, just delayed by 5-10 seconds. So my current login process is to enter my password, wait for the dots to appear in the password field in 5-10 secs, then hit return. Note that I run my MBP in clamshell mode and it's set to _never_ sleep, but instead to simply sleep the display.
2. Keyboard Entry Disappears for 10-15 Secs: About once or twice a day, keyboard input from the keyboard goes away for 5-10 seconds. This feels like the same as the first scenario above, as they do eventually appear after 5-10 seconds but happens randomly through the OS.

**Remedies Tried**
A day or so ago I updated to the latest firmware, from Nov 2, 2023. I' also removed pairing from macOS and repaired to my MBP. Alas neither of this actions have resulted in any better results. I've also tried turning my keyboard on and off to force a reconnection to the OS. This does work but about three quarters of the time I lose the right half along the way so this is more hassle than it's worth as it takes longer and more fiddling to get my right half to connect to the left versus just waiting to get past the bluetooth Bermuda Triangle.

**Additional Thoughts**
My other bluetooth peripherals, a Logi Bluetooth trackball and AirPod Pros (2nd Gen.) work flawlessly with no dropouts. But as this issue first began after the upgrade to 14.1.0, I'm not ruling out that this is a bluetooth issue/bug that Apple themselves have.

ReFil, 2023-11-05:
Hi Patrick, thank you for the bug report. If it just started happening after an update then there was probably a minor tweak on macOS that caused this. You could try changing the Bluetooth connection parameters. Remove or comment out these lines from your adv360_left_defconfig https://github.com/KinesisCorporation/Adv360-Pro-ZMK/blob/V3.0/config/boards/arm/adv360/adv360_left_defconfig#L53-L57

As for the left to right connection, after the newest update it should be really snappy, is it taking a long time to connect?

patricksalo, 2023-11-06:
Hi ReFil, as always, you're super helpful and know exactly how to solve things. I adjusted the lines mentioned and have about 5 hours under my belt and there's a night and day difference. I've seen zero lag either from the unlock dialog or any other part of the OS. Of course not 100% conclusive until I log more time, but thus far things look good and I'll report back if I find anything goes sideways.

I noticed in the config file the following:

CONFIG_BT_BAS=n #Set this to y to enable BLE battery reporting

...and although earlier today I left this just as it was, a few minutes ago I set to y and recompiled. IIRC this was turned on in builds past and I actually like this feature as it reports out my keyboard power (I think just left) to [AirBuddy](https://v2.airbuddy.app).

**Any ill effects I might see as a result of turning this on other than maybe slightly degraded battery life?**

Per the left to right connection issue, I spent some time deciphering my experience and I think _we're both correct_. I've historically turned each half off/on simultaneously per the original shipping instructions and this appears to be what causes the right keyboard to blink red most of the time under the 3.x firmwares.

But if I readjust my workflow to turn on my left half first, wait a second or two, then turn on the right half, things are super solid and always connect. So I think you're right, this is much improved and I haven't seen any failures in connection in this method. But mind you, I was just trying to replicate my experience here as pretty much the only time I ever turn off/on the halves is after a firmware update, otherwise I keep them on all the time.

Thanks again for the help.

ReFil, 2023-11-06:
Hi Patrick

Glad to hear it's all working out for you now, if it works for a bit longer with no hassle is it alright if you close this issue out?

`CONFIG_BT_BAS=y` can cause Mac's to wake-up from sleep when the battery report is set, but I guess if your laptop never goes to sleep you can enable that without issue

patricksalo, 2023-11-06:
Hi ReFil, I really wanted to see what the overnight experience was after the firmware changes and I can confirm things remain just a good as I originally reported. Yes, I think we can close this one out. Also thanks for the background on _Direct Advertising_ as this is very interesting and definitely explains the situation.