KinesisCorporation/Adv360-Pro-ZMK #266: check firmware version

Acetolyne, 2023-11-14:
Hi,
Currently when you flash new firmware I cant find a way to tell if the firmware was successfully updated. 
I have tried to run a diff on the files on the keyboard and the ones outputted from the build but there is some binary difference
The PRO version also does not have the "status report" functionality that the wired version has so I cant find a way to output the version of firmware that is currently running or verify the version in any other way.

Is there a known way to check the firmware version currently running on the 369 PRO version?
Thanks for any feedback on this. I have been unable to locate any documentation on how to achieve this.

ReFil, 2023-11-14:
Right now this is not possible, but this is several new solutions in mind to improve this

KinesisCorporation, 2023-11-14:
One "hack" is to add a custom mapping somewhere, or you could add macro in the mod layer that has a date code or other unique build identifier. We may even look to do that at the factory.

metageek, 2023-11-14:
Is there even any way to tell whether my keyboard shipped with 2.0 or 3.0? I need to know whether to run the 2.0 settings reset before applying my own build.

Acetolyne, 2023-11-14:
Thx for the information.  This would be a great improvement in the future but for now I'll work on an unsupported way of dealing with this. Appreciate everyone's input

closed 2023-11-14 by Acetolyne

ReFil, 2023-11-15:
https://github.com/KinesisCorporation/Adv360-Pro-ZMK/pull/267