KinesisCorporation/Adv360-Pro-ZMK #271: Mod layer causing build failure?

simplemachine92, 2023-11-18:
Not sure what the particular reason is but I just deleted the mod layer and the build passes fine immediately after... but heres a trace

[image]

ReFil, 2023-11-18:
Thanks for bringing this to our attention, it looks like the keymap editor and the new versioning system are conflicting, I will rectify this asap. As a mitigation you can remove the ver macro from the mod layer and builds will succeed again

ReFil, 2023-11-18:
Rectified now: https://github.com/KinesisCorporation/Adv360-Pro-ZMK/commit/d0ffaf89f8f2cd985b7750bcfeee1feaa5d7435c