KinesisCorporation/Adv360-Pro-ZMK #273: https://kinesiscorporation.github.io/Adv360-Pro-GUI auto-generates incompatible code with V3.0

trein, 2023-11-18:
Steps to reproduce the issue:

- Access https://kinesiscorporation.github.io/Adv360-Pro-GUI.
- Change any key mapping and commit.
- Example of change: https://github.com/trein/Adv360-Pro-ZMK/commit/567b0f68e9ebb9aa9f2ee1c7775ee1d97943236a

Errors:

- Github actions fail to compile the firmware.

```
Run west build -s zmk/app -d build/left -b adv360_left -- -DZMK_CONFIG="${GITHUB_WORKSPACE}/config"
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/zmk/app
-- Adding ZMK config directory as board root: /__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/config
-- ZMK Config directory: /__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/config
-- Using keymap file: /__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/config/adv360_left.keymap
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.10") found components: Interpreter 
-- Cache files will be written to: /__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/zephyr/.cache
-- Zephyr version: 3.2.0 (/__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/zephyr)
-- Found west (found suitable version "0.14.0", minimum required is "0.7.1")
-- Board: adv360_left
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.15.2 (/opt/zephyr-sdk-0.15.2)
-- Found toolchain: zephyr 0.15.2 (/opt/zephyr-sdk-0.15.2)
-- Found Dtc: /opt/zephyr-sdk-0.15.2/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/config/boards/arm/adv360/adv360_left.dts
-- Found devicetree overlay: /__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/config/adv360_left.keymap
devicetree error: /keymap/layer_mod: undefined node label 'macro_ver'
-- In: /__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/build/left/zephyr, command: /usr/bin/python3.8;/__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/zephyr/scripts/dts/gen_defines.py;--dts;/__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/build/left/zephyr/zephyr.dts.pre;--dtc-flags;'';--bindings-dirs;/__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/zmk/app/dts/bindings;/__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/zmk/app/module/dts/bindings;/__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/zephyr/dts/bindings;--header-out;/__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/build/left/zephyr/include/generated/devicetree_generated.h.new;--dts-out;/__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/build/left/zephyr/zephyr.dts.new;--edt-pickle-out;/__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/build/left/zephyr/edt.pickle;--vendor-prefixes;/__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/zmk/app/dts/bindings/vendor-prefixes.txt;--vendor-prefixes;/__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/zephyr/dts/bindings/vendor-prefixes.txt
CMake Error at /__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/zephyr/cmake/modules/dts.cmake:[23](https://github.com/trein/Adv360-Pro-ZMK/actions/runs/6914647454/job/18812708759#step:9:24)0 (message):
  gen_defines.py failed with return code: 1
Call Stack (most recent call first):
  /__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/zephyr/cmake/modules/zephyr_default.cmake:108 (include)
  /__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:15 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/build/left -GNinja -DBOARD=adv360_left -DZMK_CONFIG=/__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/config -S/__w/Adv360-Pro-ZMK/Adv360-Pro-ZMK/zmk/app
Error: Process completed with exit code 1.
```

ReFil, 2023-11-18:
Thanks for bringing this to my attention, the problem has been rectified now: https://github.com/KinesisCorporation/Adv360-Pro-ZMK/commit/d0ffaf89f8f2cd985b7750bcfeee1feaa5d7435c

closed 2023-11-18 by ReFil

trein, 2023-11-18:
Thanks for the quick turnaround!