KinesisCorporation/Adv360-Pro-ZMK #275: Issue with builds using Adv360-Pro-GUI V3.0

JamyGolden, 2023-11-19:
I can't seem to get V3.0 to build when I add a `Updated keymap` commit using Adv360-Pro-GUI. I've just [changed some `&kp`  values](https://github.com/JamyGolden/Adv360-Pro-ZMK/commits/V3.0) on the base layer and those changes don't build. It looks like this issue was dealing with the same thing: https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/271

I'm building with the latest [`ver_macro` revert commit included](https://github.com/JamyGolden/Adv360-Pro-ZMK/commit/d0ffaf89f8f2cd985b7750bcfeee1feaa5d7435c) but I'm still getting the error: https://github.com/JamyGolden/Adv360-Pro-ZMK/actions/runs/6923609888

ReFil, 2023-11-20:
Hiya looking at your build failure the problem isn't connected to that, it seems that there are two macro_kinesis nodes defined https://github.com/JamyGolden/Adv360-Pro-ZMK/actions/runs/6923609888/job/18831714924#step:9:21

ReFil, 2023-11-20:
https://github.com/JamyGolden/Adv360-Pro-ZMK/commit/10d48753bba571961ada7db6c26580d4b65b11d4#diff-cce988ab109aac914e6a52deacdc25e43e1188e4e79ef9f97103a466a84918caR31

Down here it looks like the keymap editor doubled up on the macros. If you edit macros.dtsi to remove one of the two macro definitions it should be good

JamyGolden, 2023-11-22:
Thanks, that solved it :smile:

closed 2023-11-22 by JamyGolden