KinesisCorporation/Adv360-Pro-ZMK #277: Constant bluetooth disconnects after upgrading to v3.0

jamesgour, 2023-11-27:
Hey Kinesis team,

I upgraded last week from the v2.0 branch to v3.0. Before doing the upgrade, I even flashed BOTH reset files to make sure things would be clean.

Unfortunately, across 2 different computers I am seeing constant bluetooth disconnects. Very randomly the keyboard will disconnect, and then struggle to reconnect for several minutes.

Is anyone else seeing these issues with the new branch? Any ideas on what I can try? 

Thank you very much for any help!

ReFil, 2023-11-27:
What operating system are you using?

jamesgour, 2023-11-27:
@ReFil Both systems are running Windows 11

JuanPabloMF, 2023-11-27:
Same here, with OS X High Sierra. And can't connect also to Pop OS (ubuntu 22.04)

ReFil, 2023-11-27:
@jamesgour are they both using intel wireless cards by any chance. There's a known issue with the latest Intel driver update breaking things

ReFil, 2023-11-27:
@JuanPabloMF this is likely a different issue. Something you can try is removing these lines from adv360_left_defconfig https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/262#issuecomment-1793689158 . Please also ensure that the Bluetooth profile led is fast flashing and the host you're trying to pair to hasn't been paired to another profile on the keyboard.

JuanPabloMF, 2023-11-27:
> @JuanPabloMF this is likely a different issue. Something you can try is removing these lines from adv360_left_defconfig [#262 (comment)](https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/262#issuecomment-1793689158) . Please also ensure that the Bluetooth profile led is fast flashing and the host you're trying to pair to hasn't been paired to another profile on the keyboard.

Just tried and updated my keyboard, and it didn't work (not with Pop OS nor with OS X)

ReFil, 2023-11-27:
@JuanPabloMF please open a new issue for your problem

jamesgour, 2023-11-27:
@ReFil Okay I will try that fix when I get home, thanks for the idea :).

If it doesn't work, is the guidance to just go back to the v2.0 firmware? I never had any issues when I was using that.

Also, is there any ETA on when the fixes will be in place if the issue is coming from the intel wireless card?

ReFil, 2023-11-27:
You should be able to revert the merge commit, we keep v2.0 base branch online so users can still build changrs

jamesgour, 2023-11-27:
Okay thanks! I'll keep an eye out for when things even out in v3.0 :)

closed 2023-11-27 by jamesgour

ReFil, 2023-11-27:
Thank you for your understanding. One last thing that might be worth trying is manually clearing each Bluetooth profile of any residual information using mod+1-5 and mod+win on the stock layout

We're working hard to improve the Bluetooth experience with continual firmware updates and improvements. If there's any major changes that have the potential to improve the experience with windows I will reply to this issue again

ReFil, 2023-11-28:
Looking at your repo you still have v2.0 branch setup and available, if you change to pushing to this branch you can edit things as per normal

https://github.com/jamesgour/Adv360-Pro-ZMK/tree/V2.0

jamesgour, 2023-11-28:
@ReFil Yes, I rolled back to v2.0 using an old commit I had from back then. 

Unfortunately, after rolling back now the keyboard won't even connect to my Windows 11 machine (Intel Bluetooth / Intel Wireless card). 

I keep getting "Try connecting your device again".

I was thinking I should update my v2.0 branch to try the final commit to see if it fixes this new issue?

ReFil, 2023-11-28:
Ah I see, it shouldn't make a difference because the Bluetooth code doesn't get updated with the config repo. Please double check every profile has had the residual pairing information cleared out as it definitely seems like that could be the issue

jamesgour, 2023-11-28:
@ReFil All BT profiles cleared and still no love, unfortunately. It's odd because I never had pairing issues before on that old branch, but perhaps the newer BT drivers have caused something.

Have you seen this "Try connecting your device again" issue before? I'm debating getting another bluetooth adapter dongle or something but it would be really nice to at least connect after all of this headache.

ReFil, 2023-11-28:
The "couldn't connect, try connecting again" error is indicative of the board rejecting the pairing attempt from the pc, which can happen for a few different reasons but is usually a security key exchange problem

jamesgour, 2023-11-28:
@ReFil Hmmm okay any other ideas on what to try to resolve a security key exchange issue? 

Or do you think I need to just try another Bluetooth adapter?

ReFil, 2023-11-28:
Have you tried pairing it to a phone or some other Bluetooth device to ensure the keyboard is working properly?

jamesgour, 2023-11-28:
Good idea, yes I just paired it to my phone and I was able to pair it to my other computer running windows 11. That computer does not have an Intel wireless card though.