KinesisCorporation/Adv360-Pro-ZMK #278: Bluetooth connects and inmediately disconnects with Pop OS and OS X

JuanPabloMF, 2023-11-27:
Just got my kinesis 360 pro (beginning of november). I have tried connecting bluetooth to my linux machine (Pop OS 22.04) and my OS X laptop (High Sierra), but with both I manage to connect and inmediately loose connection. Also with the OS X laptop the machine enters a loop of retrys for connection and a window pops up every 2 seconds.

I have tried for fixing:

- Poiting west.yml to 3.2-beta revision.
- Commenting lines 53 to 57 in left-defconfig as suggested by ReFil

ReFil, 2023-11-27:
What Bluetooth adapter is in use on your Linux machine and what MacBook are you using? Is it definitely Bluetooth LE compatible?

JuanPabloMF, 2023-11-28:
Looks like both have:

```
btmgmt info
Index list with 1 item
hci0:	Primary controller
	addr 44:AF:28:10:6E:67 version 11 manufacturer 2 class 0x7c0104
	supported settings: powered connectable fast-connectable discoverable bondable link-security ssp br/edr hs le advertising secure-conn debug-keys privacy configuration static-addr phy-configuration wide-band-speech 
	current settings: powered bondable ssp br/edr le secure-conn 
	name pop-os
	short name 
```

Mabook Pro is from 2017 and since 2012 they should have support.

JuanPabloMF, 2023-11-28:
It worked with a windows laptop. OS X still with the connection, disconnection loop. Trying to add a bluetooth dongle to the linux machine to see if it works.

ReFil, 2023-11-28:
Pop OS is Ubuntu based right? My personal computer is also Ubuntu based and has been very solid with the 360 pro. On the pop OS machine does it also go between connected and disconnected rapidly?

JuanPabloMF, 2023-11-28:
Yes ubuntu based. It just connects and then inmediately disconnects once (os x gets into a loop). It work fine with my Ipad also. But I'm trying another wifi/bt antenna for pop os cause I have my doubts on that hardware part (tried a speaker and signal was not very clear).

ReFil, 2023-11-28:
Just to confirm, you are using a new clear profile for each new device you are trying to pair to? If you try and pair the same device to two different profiles or unpair on one device but not on the keyboard it will not pair or connect to that device

JuanPabloMF, 2023-11-28:
Yes I tried clearing profiles.

JuanPabloMF, 2023-11-30:
I made it work in Pop! OS by using a BT dongle instead of the built-in card. No results yet with OS X.

ReFil, 2023-11-30:
Good to hear it's now working on Pop OS, unfortunately I don't have a mac running high Sierra, it's quite an old version, are you able to update the Mac at all?

nkane, 2023-12-15:
I've had a similar experience as well, I'm running Ubuntu on a new System76 machine: 
```
➜  ~ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.3 LTS
Release:        22.04
Codename:       jammy
➜  ~ btmgmt info
Index list with 1 item
hci0:   Primary controller
        addr B0:DC:EF:33:4D:FB version 12 manufacturer 2 class 0x7c010c
        supported settings: powered connectable fast-connectable discoverable bondable link-security ssp br/edr hs le advertising secure-conn debug-keys privacy configuration static-addr phy-configuration wide-band-speech
        current settings: powered connectable discoverable bondable ssp br/edr le secure-conn
        name system76-pc
        short name
hci0:   Configuration options
        supported options: public-address
        missing options:
```

For some reason, whenever I just spam `pair` request in `bluetoothctl` it eventually is able to connect even though it was marked as previously paired.
```
[bluetooth]# trust F7:83:49:12:61:C6
Changing F7:83:49:12:61:C6 trust succeeded
[CHG] Device F7:83:49:12:61:C6 RSSI: -68                          
[NEW] Device B0:E4:D5:0D:2C:41 B0-E4-D5-0D-2C-41
[bluetooth]# pair F7:83:49:12:61:C6                                    
Attempting to pair with F7:83:49:12:61:C6
[CHG] Device F7:83:49:12:61:C6 Connected: yes                     
[CHG] Device F7:83:49:12:61:C6 Connected: no
Failed to pair: org.bluez.Error.AuthenticationCanceled
[bluetooth]# pair F7:83:49:12:61:C6                                                                                                         
Attempting to pair with F7:83:49:12:61:C6                                                                                                   
[CHG] Device F7:83:49:12:61:C6 Connected: yes                                                                                               
[CHG] Device F7:83:49:12:61:C6 Connected: no                                                                                                
Failed to pair: org.bluez.Error.AuthenticationCanceled
[bluetooth]# pair F7:83:49:12:61:C6                                   
Attempting to pair with F7:83:49:12:61:C6                         
[bluetooth]# pair F7:83:49:12:61:C6                                   
Attempting to pair with F7:83:49:12:61:C6       
Failed to pair: org.bluez.Error.InProgress
[CHG] Device F7:83:49:12:61:C6 Connected: yes
[CHG] Device F7:83:49:12:61:C6 Connected: no    
Failed to pair: org.bluez.Error.AuthenticationCanceled
[bluetooth]# pair F7:83:49:12:61:C6                                   
Attempting to pair with F7:83:49:12:61:C6
[CHG] Device F7:83:49:12:61:C6 Connected: yes
[CHG] Device F7:83:49:12:61:C6 Paired: yes
Pairing successful 
```

ReFil, 2023-12-15:
Are you using an intel bluetooth controller?

nkane, 2023-12-15:
I'm using this laptop: https://system76.com/laptops/darter  

Not entirely sure how to check if it's an intel bluetooth controller, this is all that `bluetoothctl` is showing me: 

```
[Adv360 Pro]# show
Controller B0:DC:EF:33:4D:FB (public)
        Name: system76-pc
        Alias: system76-pc
        Class: 0x007c010c
        Powered: yes
        Discoverable: yes
        DiscoverableTimeout: 0x000000b4
        Pairable: yes
        UUID: Message Notification Se.. (00001133-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: OBEX Object Push          (00001105-0000-1000-8000-00805f9b34fb)
        UUID: Message Access Server     (00001132-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        UUID: IrMC Sync                 (00001104-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (00005005-0000-1000-8000-0002ee000001)
        UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Phonebook Access Server   (0000112f-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree Audio Gateway   (0000111f-0000-1000-8000-00805f9b34fb)
        UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
        UUID: OBEX File Transfer        (00001106-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v1D6Bp0246d0540
        Discovering: yes
        Roles: central
        Roles: peripheral
Advertising Features:
        ActiveInstances: 0x00 (0)
        SupportedInstances: 0x0c (12)
        SupportedIncludes: tx-power
        SupportedIncludes: appearance
        SupportedIncludes: local-name
        SupportedSecondaryChannels: 1M
        SupportedSecondaryChannels: 2M
        SupportedSecondaryChannels: Coded
```

I'm guessing it is, because of this: 
```
➜  ~ sudo dmesg | grep -i Bluetooth
[    1.844935] Bluetooth: Core ver 2.22
[    1.844958] NET: Registered PF_BLUETOOTH protocol family
[    1.844959] Bluetooth: HCI device and connection manager initialized
[    1.844962] Bluetooth: HCI socket layer initialized
[    1.844964] Bluetooth: L2CAP socket layer initialized
[    1.844967] Bluetooth: SCO socket layer initialized
[    1.995226] Bluetooth: hci0: Device revision is 0
[    1.995237] Bluetooth: hci0: Secure boot is enabled
[    1.995240] Bluetooth: hci0: OTP lock is enabled
[    1.995243] Bluetooth: hci0: API lock is enabled
[    1.995245] Bluetooth: hci0: Debug lock is disabled
[    1.995248] Bluetooth: hci0: Minimum firmware build 1 week 10 2014
[    1.995253] Bluetooth: hci0: Bootloader timestamp 2019.40 buildtype 1 build 38
[    2.010890] Bluetooth: hci0: Found device firmware: intel/ibt-0040-0041.sfi
```

ReFil, 2023-12-15:
Please try the setting mentioned here: https://github.com/zmkfirmware/zmk/issues/1487#issuecomment-1836591238 in `adv360_left_defconfig`

ReFil, 2023-12-15:
Darter Pro uses an intel AX201 it seems: https://tech-docs.system76.com/models/darp8/repairs.html#replacing-the-wireless-card

nkane, 2023-12-15:
@ReFil this seems to have kind of resolved the issue, but now the right side keyboard will randomly not be able to connect to the left side. All I did was set `CONFIG_BT_CTLR_PHY_2M=n` in the left config like suggested in that issue. There were some other settings I can give a shot if I keep having issues with the right side not picking up the left.

nkane, 2023-12-15:
Unfortunately, after power cycling the keyboards at the end of the day the right side wasn't able to connect to the left side with the following added to the `adv360_left_defconfig`: 
```
CONFIG_BT_CTLR_PHY_2M=n
CONFIG_BT_GATT_AUTO_SEC_REQ=y
```
Tried it with and without  `CONFIG_BT_GATT_AUTO_SEC_REQ=y`, and it made no difference.

ReFil, 2023-12-15:
Try adding this to left and right defconfigs `CONFIG_BT_LL_SW_LLCP=y`

bryant-bourgeois, 2023-12-16:
So, I was having a similar rapid disconnect/reconnect problem where I was typically unable to get a working connection on my System76/PopOS laptop (no issues on my macbooks though).

After adding the below lines to the left_defconfig file:
CONFIG_BT_CTLR_PHY_2M=n
CONFIG_BT_LL_SW_LLCP=y

And this line to the right_defconfig file:
CONFIG_BT_LL_SW_LLCP=y

Things seem to be working fine after flashing and a re-pair to the System76 machine.

JuanPabloMF, 2023-12-29:
> Good to hear it's now working on Pop OS, unfortunately I don't have a mac running high Sierra to test with, it's quite an old version, are you able to update the Mac at all?

Updated to Big Sur, still facing same problem, will try this kind of fix:

> After adding the below lines to the left_defconfig file:
> CONFIG_BT_CTLR_PHY_2M=n
> CONFIG_BT_LL_SW_LLCP=y
> 
> And this line to the right_defconfig file:
> CONFIG_BT_LL_SW_LLCP=y

ReFil, 2023-12-29:
Thanks for the feedback, At this point i'm out of ideas unfortunately, it seems like it just doesn't want to work with your mac

ReFil, 2024-01-07:
@JuanPabloMF  We have developed a new beta firmware that has some changes in configuration that might improve the Mac issue affecting you. If you would be able to try it and provide feedback it would be much appreciated:  [360pro-beta-1-6-24.zip](https://github.com/KinesisCorporation/Adv360-Pro-ZMK/files/13854902/360pro-beta-1-6-24.zip)

OleJoik, 2024-01-09:
@ReFil I just got my Adv360 Pro ZMK today, and have a fair amount of issues with the bluetooth connectivity. It seems to me like your newly developed beta firmware (1-6-24) works much better for me than the previous versions that I tried, namely 3.0 and 3.2-beta. I have not been able to use my configuration on that firmware (i tried hard...), but I assume it can't be used in west.yaml yet...?

System is recently updated Arch Linux. 

Connectivity issues seems to be the same both in beta (adv360-z3.2-beta), and in 3.0. The configuration in where I spent the most time was with this west.yaml:

`
manifest:
  remotes:
    - name: zmkfirmware
      url-base: https://github.com/zmkfirmware
    - name: refil
      url-base: https://github.com/refil
  projects:
    - name: zmk
      remote: refil
      revision: adv360-z3.2-beta
      import: app/west.yml
  self:
    path: config
`
I have tried various combinations of trust / pair / connect using my systems bluetooth tooling
typical bluetoothctl output below:
`
$ bluetoothctl
Waiting to connect to bluetoothd...[bluetooth]# Agent registered
[bluetooth]# AdvertisementMonitor path registered
[bluetooth]# [CHG] Controller ZZ:ZZ:ZZ:ZZ Discovering: no

$ scan on
[bluetooth]# SetDiscoveryFilter success
[bluetooth]# Discovery started
[bluetooth]# [CHG] Controller ZZ:ZZ:ZZ:ZZ Discovering: yes
[bluetooth]# [NEW] Device XX:XX:adv360:XX:XX Adv360 Pro

$ trust XX:XX:adv360:XX:XX
[bluetooth]# [CHG] XX:XX:adv360:XX:XX Trusted: yes
[bluetooth]# Changing XX:XX:adv360:XX:XX trust succeeded

$ pair XX:XX:adv360:XX:XX
Attempting to pair with XX:XX:adv360:XX:XX
[CHG] Device XX:XX:adv360:XX:XX Connected: yes
[CHG] Device XX:XX:adv360:XX:XX Connected: no
[bluetooth]# Failed to pair: org.bluez.Error.AuthenticationCanceled
[bluetooth]# [CHG] Device XX:XX:adv360:XX:XX RSSI: 0xffffffba (-70)
[bluetooth]# [DEL] Device XX:XX:adv360:XX:XX Adv360 Pro

$ connect XX:XX:adv360:XX:XX
Attempting to connect to XX:XX:adv360:XX:XX
[bluetooth]# [CHG] Device XX:XX:adv360:XX:XX WakeAllowed: yes
[bluetooth]# [NEW] Device YY:YY:YY:YY:YY:YY YY-YY-YY-YY-YY-YY
[CHG] Device XX:XX:adv360:XX:XX Connected: yes
[Adv360 Pro]# Connection successful
[CHG] Device XX:XX:adv360:XX:XX Connected: no
[CHG] Device XX:XX:adv360:XX:XX Connected: yes
[CHG] Device XX:XX:adv360:XX:XX Connected: no
[CHG] Device XX:XX:adv360:XX:XX Connected: yes
[CHG] Device XX:XX:adv360:XX:XX Connected: no
[CHG] Device XX:XX:adv360:XX:XX Connected: yes
[CHG] Device XX:XX:adv360:XX:XX Connected: no
[CHG] Device XX:XX:adv360:XX:XX Connected: yes
[CHG] Device XX:XX:adv360:XX:XX Connected: no
[CHG] Device XX:XX:adv360:XX:XX Connected: yes
[CHG] Device XX:XX:adv360:XX:XX Connected: no
[CHG] Device XX:XX:adv360:XX:XX Connected: yes
[CHG] Device XX:XX:adv360:XX:XX Connected: no
[CHG] Device XX:XX:adv360:XX:XX Connected: yes
[CHG] Device XX:XX:adv360:XX:XX Connected: no
[CHG] Device XX:XX:adv360:XX:XX Connected: yes

this goes on forever.......
`

At this point, it did seem like I sometimes got signals though, but typically I would press a button, and my system would get stuck on something. For example like thisssssssssssssssssssssss .... (if that makes sense)...

However, on beta-1-6-24, I have no connectivity issues at all. Would very much appreciate if you could make this available for use with my own config, so I can rid myself of the cable clutter on my desk...

If you have any questions I am happy to supply more feedback.

OleJoik, 2024-01-10:
I added the configuration changes mentioned above:

    left_defconfig file:
    CONFIG_BT_CTLR_PHY_2M=n
    CONFIG_BT_LL_SW_LLCP=y

   right_defconfig file:
    CONFIG_BT_LL_SW_LLCP=y

This seems to have resolved the issue, but I have not very thoroughly tested it yet.

ReFil, 2024-01-14:
These configuration changes will be made upstream with the merging of #326 under the config option of `CONFIG_ZMK_BLE_EXPERIMENTAL_FEATURES`. It might cause issues if users have already made the changes separately and thus when it's merged it would be a good idea to remove the `CONFIG_BT_CTLR_PHY_2M=n` and `CONFIG_BT_LL_SW_LLCP=y` lines where they occur

PR #326 (2024-01-14): Base ZMK update
## Advantage 360 Pro PR template

### What's changed:

This will move users to a new ZMK update and has the configration changes that are prerequisites to this change

### Why has this change been implemented:

There are new features and improvements to bluetooth connectivity implemented. This should resolve #278 #259 #172 and potentially #104 as well

### What (if any) actions must a user take after this change:

Users who have added `CONFIG_BT_CTLR_PHY_2M=n` and `CONFIG_BT_LL_SW_LLCP=y` for interim compatibility with Intel NICs should remove these lines from their _defconfig  [... cut]
closed 2024-01-14 by ReFil