KinesisCorporation/Adv360-Pro-ZMK #282: Bluetooth unable to connect after a few days

tbenst, 2023-12-01:
On all of my profiles (Macbook, iPad, iPhone), after a couple days I have to reset to Adv360 profiles with mod-win key as they no longer pair. I have no issues with my other bluetooth keyboards, mice, or other devices. 

Anecdotally, the issues seems to occur more often if the keyboard is left with the toggles in the ON position overnight, I haven't noticed this occurring after being in the OFF position.

Please note that power toggling does not resolve.

KinesisCorporation, 2023-12-01:
Hi Tyler,

Thanks for sharing this and sorry to hear about it. Can you try leaving the keyboard powered on exclusively for a week and report back? That shouldn't have any impact on battery life as they go to sleep after 30 seconds.

@ReFil may have some ideas on your config settings too.

tbenst, 2023-12-02:
I'm happy to report that I've figured out how to reliably reproduce the issue.

1. using keyboard with laptop on desk
2. clsoe lid, laptop moved to another room (~35ft w/ one wall), keyboard left on table (switches in on position)
3. Use laptop for a bit, wait until keyboard disconnects (~3-5 min)
4. close lid, move laptop next to keyboard on desk
5. Notice that keyboard not paired, power toggling does not fix

In order to fix I have to:
1. `Mod-windows` to reset keyboard bluetooth profile
2. delete profile on device
3. power toggle keyboard (necessary!)
4. connect as new device

After making the fix suggested by @ReFil, the issue does appear to be resolved. 

I understand that pairing challenges with bluetooth are frustrating to debug. I know hardware testing automation is one approach for testing new firmware across a complex user story of pairing, unpairing, losing signal range, reconnecting, power toggling, etc. 

Hope this fix can be deployed by default!

ReFil, 2023-12-03:
If the issue hasn't recurred would you be able to close this GitHub issue out as solved? Many thanks

ReFil, 2023-12-05:
Hi Tyler, just following up, has the issue recurred?

tbenst, 2023-12-05:
The issue has not occurred on my keyboard in the past few days.

Perhaps it would make sense to close this issue with a commit to the kinesis repo that makes this change for everyone?

If this simple fix solves the underlying issue, and I think it does, it'll make lots of folks happy, and they shouldn't have to manually adjust these settings.

https://www.reddit.com/r/kinesisadvantage/comments/1183xdh/hey_has_anyone_had_connection_issues_i_cant_get/

https://www.reddit.com/r/kinesisadvantage/comments/12z30xz/advantage360_pro_bluetooth_connectivity_problems/

ReFil, 2023-12-05:
Those settings are planning to be merged in a upcoming update

ReFil, 2023-12-06:
Settings are now default as of #289

closed 2023-12-06 by ReFil

ReFil, 2023-12-07:
@humblehacker Sorry to bother you, I noticed you have removed some lines from your defconfig file to mirror some changes however the link pointed to some lines that have since been removed so the reference is no longer valid and you have inadvertently removed some other lines. In this commit: https://github.com/KinesisCorporation/Adv360-Pro-ZMK/commit/2055aa2498ee7749f237a31f317ec1d985c377bb the changes were merged into the official firmware so you don't have to make other alterations. My apologies for any confusion caused

humblehacker, 2023-12-07:
Talk about proactive customer service! Thanks, I was concerned that the lines may have been out of sync since the link pointed to a branch and not a specific commit.  Surprisingly, the keyboard is working with those changes on bluetooth again (had it plugged in for quite a while). So thanks for the heads up. I'll revert my changes.

tbenst, 2023-12-21:
Hey! Unfortunately it seems that this fix doesn't fully fix the issue--if I increase the time in which the laptop is apart from the keyboard, the bluetooth profile ends up in a broken state