KinesisCorporation/Adv360-Pro-ZMK #286: rsync:

lsnizek, 2023-12-05:
I am having trouble downloading firmware. My custom build which used to download fine is now returning:

```
rsync: rename "/Volumes/ADV360PRO/.left.uf2.y8fl7o" -> "left.uf2": No such file or directory (2)
rsync error: some files could not be transferred (code 23) at /AppleInternal/Library/BuildRoots/11aa8fb2-5f4b-11ee-bc7f-926038f30c31/Library/Caches/com.apple.xbs/Sources/rsync/rsync/main.c(996) [sender=2.6.9]
```

Same firmware build, same MacBook hardware. The only thing changed that I can think of is macOS, now Sonoma 14.1.1, previously Ventura.

In addition, when I move up to the latest V3.0 branch with the build filename changes of cdcec22, I get a different error:

```
rsync: write failed on "/Volumes/ADV360PRO/202312051056-1511c89-left.uf2": Input/output error (5)
rsync error: error in file IO (code 11) at /AppleInternal/Library/BuildRoots/11aa8fb2-5f4b-11ee-bc7f-926038f30c31/Library/Caches/com.apple.xbs/Sources/rsync/rsync/receiver.c(268) [receiver=2.6.9]
rsync: connection unexpectedly closed (52 bytes received so far) [generator]
rsync error: error in rsync protocol data stream (code 12) at /AppleInternal/Library/BuildRoots/11aa8fb2-5f4b-11ee-bc7f-926038f30c31/Library/Caches/com.apple.xbs/Sources/rsync/rsync/io.c(453) [generator=2.6.9]
rsync: connection unexpectedly closed (34 bytes received so far) [sender]
rsync error: error in rsync protocol data stream (code 12) at /AppleInternal/Library/BuildRoots/11aa8fb2-5f4b-11ee-bc7f-926038f30c31/Library/Caches/com.apple.xbs/Sources/rsync/rsync/io.c(453) [sender=2.6.9]
```

My procedure has been to put module into bootloader mode (eg Mod+HK1) and run `rsync left.uf2 /Volumes/ADV360PRO` using files from generated by the GitHub CI

What I am trying to do is tweak a key mapping

ReFil, 2023-12-05:
Have you tried just copying the uf2 files using finder?

lsnizek, 2023-12-05:
> Have you tried just copying the uf2 files using finder?

I have, get a similar sounding error in a pop-up window

ReFil, 2023-12-05:
Are tha changes actually propagated to the board? It is not uncommon for finder to throw an error but the changes actually do get flashed anyway

lsnizek, 2023-12-05:
> Are tha changes actually propagated to the board? It is not uncommon for finder to throw an error but the changes actually do get flashed anyway

Yes they are. It works with Finder or rsync. Thank you so much.

Here is the Finder error, for completeness

[image]