KinesisCorporation/Adv360-Pro-ZMK #292: Bluetooth not connecting after latest firmware flash

sansari, 2023-12-09:
I just flashed my Kinesis 360 Pro to the [latest](https://github.com/KinesisCorporation/Adv360-Pro-ZMK/commit/11932126d4b99fb1d2f0bfecfb138dd7adcec3b6) firmware as of today. Upon doing so, I noticed two issues:

1. **Flashing Error:** The flashing process seems to now show more errors than before. It still succeeds, but it looks like a failure and not particularly comforting to see this every time. Was this change intentional?

[image]

2. **Bluetooth No Longer Connects:** I can't get the keyboard to connect over Bluetooth anymore. I see the keyboard show up in the Bluetooth options, but tapping it just makes a spinner appear next to it and then it never actually connects. The user profile light indicating it's not connected to Bluetooth just keeps flashing. I've tried restarting computer, unplug/replugging, and of course I had just flashed the keyboards so they've been reset.

I'm on an M2 Macbook Air with macOS Sonoma 14.1.2.

Thanks for your help!

ReFil, 2023-12-09:
Hi, thanks for reporting these issues and I'm sorry you're having problems getting it to connect. 

1: this is not the fault of the board, this is a known artifact of the newer macOS updates, unfortunately nothing can be done about this

2: is the Bluetooth led flashing slowly or quickly, by "reset" do you mean you flashed the "settings-reset.uf2" file onto it?

sansari, 2023-12-09:
Thanks for the fast response! Re: #2, it seems to be flashing slowly, it's set on profile 1 but if I set it to 2 (with Mod+2) it flashes quickly. Does that mean it's previously connected on 1 but is unable to connect right now?

By reset I actually just meant I updated the configuration (i.e. replaced `left.uf2` and `right.uf2`). I did not do a full settings reset.

ReFil, 2023-12-09:
Ok, since some Bluetooth settings changed in this recent update can I suggest pressing mod+windows to unpair on the keyboard, unpairing the keyboard on your computer and trying to pair it again?

sansari, 2023-12-09:
Can you clarify which key is "windows" in this case? I'm on a Mac, so do you mean Command? I did try hitting that, disconnecting and reconnecting, same issue. Basically when you hit "Connect" it just tries and fails.

[image]

ReFil, 2023-12-09:
Please hold down key highlighted in green and then tap key highlighted in blue, the bluetooth light should start flashing rapidly
[image]

sansari, 2023-12-09:
Thank you! That did it. Bluetooth is working again 🎉 I have my Delete key remapped to Command, so I was trying to use that one, but I'm realizing that when hitting in combo with Mod, you have to use the original Windows key location.

closed 2023-12-09 by sansari

ReFil, 2023-12-09:
Glad to hear it's all working now :)

lbtrace, 2024-03-22:
> I just flashed my Kinesis 360 Pro to the [latest](https://github.com/KinesisCorporation/Adv360-Pro-ZMK/commit/11932126d4b99fb1d2f0bfecfb138dd7adcec3b6) firmware as of today. Upon doing so, I noticed two issues:
> 
> 1. **Flashing Error:** The flashing process seems to now show more errors than before. It still succeeds, but it looks like a failure and not particularly comforting to see this every time. Was this change intentional?
> 
> ![CleanShot 2023-12-09 at 14 10 50@2x](https://private-user-images.githubusercontent.com/105090/289316033-9239923d-8cf3-4995-98dd-cf62fa2ddb9d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTEwOTI0OTUsIm5iZiI6MTcxMTA5MjE5NSwicGF0aCI6Ii8xMDUwOTAvMjg5MzE2MDMzLTkyMzk5MjNkLThjZjMtNDk5NS05OGRkLWNmNjJmYTJkZGI5ZC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzIyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMyMlQwNzIzMTVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wY2RhMTQxNmIwOTkzMjVhYWE5OGE1YzQwZjFlNmIyNjU0MWM5ZWIwZGYzMWI1MDdjNGMxOWMwOGM0NDQ1NDMxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.-kvuY33A6wKbWA2aD1V5fTB-4WYmziwnsopmK4KteGU)
> 
> 2. **Bluetooth No Longer Connects:** I can't get the keyboard to connect over Bluetooth anymore. I see the keyboard show up in the Bluetooth options, but tapping it just makes a spinner appear next to it and then it never actually connects. The user profile light indicating it's not connected to Bluetooth just keeps flashing. I've tried restarting computer, unplug/replugging, and of course I had just flashed the keyboards so they've been reset.
> 
> I'm on an M2 Macbook Air with macOS Sonoma 14.1.2.
> 
> Thanks for your help!

@sansari how do you resolve issue 1? I meet same issue can't flash firmware

KinesisCorporation, 2024-03-22:
That is a "faux" error from macOS because the bootloader drive automatically disconnects after completion. The firmware is still being installed. Use the Mod + V macro to print to screen your version number to confirm.

MrMoovf, 2024-09-10:
This thread saved me! I was having the exact same issue. Currently writing happily from my newly remapped keyboard!