KinesisCorporation/Adv360-Pro-ZMK #306: Temporarily turn back on ZMK USB logging

alexandregressier, 2023-12-27:
Hello there,

I'm currently working on adding a custom behavior to a custom ZMK firmware and I need to enable logging for debugging purposes, which is now disabled by default (see https://github.com/KinesisCorporation/Adv360-Pro-ZMK/pull/101).

Here's what I've tried so far:

1. Adding back these two lines in both `adv360_left_defconfig` and `adv360_right_defconfig`.
```
CONFIG_LOG=y
CONFIG_ZMK_USB_LOGGING=y
```

2. Adding back the `&uart0` block in `adv360.dtsi` and adapt `&usbd` consequently:
```
&usbd {
    status = "okay";
    cdc_acm_uart: cdc_acm_uart {
        compatible = "zephyr,cdc-acm-uart";
        label = "CDC_ACM_0";
    };
};

&uart0 {
    compatible = "nordic,nrf-uarte";
    status = "okay";
    current-speed = <115200>;
    tx-pin = <39>;
    rx-pin = <44>;
    rts-pin = <34>;
    cts-pin = <9>;
};
```

3. Defining `zephyr,console` and an alias for `&uart0` at the top of `adv360.dtsi`:
```
model = "Adv360";
compatible = "kinesis,adv360";

aliases {
    uart0 = &uart0;
};

chosen {
    zephyr,code-partition = &code_partition;
    zephyr,sram = &sram0;
    zephyr,flash = &flash0;
    zmk,kscan = &kscan0;
    zmk,backlight = &backlight;
    zmk,battery = &vbatt;
    zmk,matrix_transform = &default_transform;
    zmk,underglow = &led_strip;
    zephyr,console = &cdc_acm_uart;
};
```

I've encountered a bunch of issues in my trial and error process, the most recent ones being related to pin control. Here's the one I'm currently stuck on:
```
[284/352] Building C object zephyr/drivers/serial/CMakeFiles/drivers__serial.dir/uart_nrfx_uarte.c.obj
FAILED: zephyr/drivers/serial/CMakeFiles/drivers__serial.dir/uart_nrfx_uarte.c.obj 
ccache /opt/zephyr-sdk-0.15.2/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DKERNEL -DNRF52840_XXAA -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR_SUPERVISOR__ -D__ZEPHYR__=1 -I/__w/kinesis-advantage360-zmk/kinesis-advantage360-zmk/zephyr/include -I/__w/kinesis-advantage360-zmk/kinesis-advantage360-zmk/build/left/zephyr/include/generated -I/__w/kinesis-advantage360-zmk/kinesis-advantage360-zmk/zephyr/soc/arm/nordic_nrf/nrf52 -I/__w/kinesis-advantage360-zmk/kinesis-advantage360-zmk/zephyr/soc/arm/nordic_nrf/common/. -I/__w/kinesis-advantage360-zmk/kinesis-advantage360-zmk/zephyr/subsys/bluetooth/controller/ll_sw/nordic/hal/nrf5/nrfx_glue -I/__w/kinesis-advantage360-zmk/kinesis-advantage360-zmk/zephyr/subsys/bluetooth -I/__w/kinesis-advantage360-zmk/kinesis-advantage360-zmk/zephyr/subsys/usb/device -I/__w/kinesis-advantage360-zmk/kinesis-advantage360-zmk/zephyr/subsys/settings/include -I/__w/kinesis-advantage360-zmk/kinesis-advantage360-zmk/modules/hal/cmsis/CMSIS/Core/Include -I/__w/kinesis-advantage360-zmk/kinesis-advantage360-zmk/modules/hal/nordic/nrfx -I/__w/kinesis-advantage360-zmk/kinesis-advantage360-zmk/modules/hal/nordic/nrfx/drivers/include -I/__w/kinesis-advantage360-zmk/kinesis-advantage360-zmk/modules/hal/nordic/nrfx/mdk -I/__w/kinesis-advantage360-zmk/kinesis-advantage360-zmk/zephyr/modules/hal_nordic/nrfx/. -I/__w/kinesis-advantage360-zmk/kinesis-advantage360-zmk/modules/crypto/tinycrypt/lib/include -I/__w/kinesis-advantage360-zmk/kinesis-advantage360-zmk/zmk/app/module/include -I/__w/kinesis-advantage360-zmk/kinesis-advantage360-zmk/zmk/app/module/drivers/sensor/battery/. -isystem /__w/kinesis-advantage360-zmk/kinesis-advantage360-zmk/zephyr/lib/libc/minimal/include -isystem /opt/zephyr-sdk-0.15.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/include -isystem /opt/zephyr-sdk-0.15.2/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/include-fixed -fno-strict-aliasing -Os -imacros /__w/kinesis-advantage360-zmk/kinesis-advantage360-zmk/build/left/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfp16-format=ieee --sysroot=/opt/zephyr-sdk-0.15.2/arm-zephyr-eabi/arm-zephyr-eabi -imacros /__w/kinesis-advantage360-zmk/kinesis-advantage360-zmk/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/__w/kinesis-advantage360-zmk/kinesis-advantage360-zmk/zmk/app=CMAKE_SOURCE_DIR -fmacro-prefix-map=/__w/kinesis-advantage360-zmk/kinesis-advantage360-zmk/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/__w/kinesis-advantage360-zmk/kinesis-advantage360-zmk=WEST_TOPDIR -ffunction-sections -fdata-sections -std=c99 -nostdinc -Wfatal-errors -MD -MT zephyr/drivers/serial/CMakeFiles/drivers__serial.dir/uart_nrfx_uarte.c.obj -MF zephyr/drivers/serial/CMakeFiles/drivers__serial.dir/uart_nrfx_uarte.c.obj.d -o zephyr/drivers/serial/CMakeFiles/drivers__serial.dir/uart_nrfx_uarte.c.obj -c /__w/kinesis-advantage360-zmk/kinesis-advantage360-zmk/zephyr/drivers/serial/uart_nrfx_uarte.c
In file included from /__w/kinesis-advantage360-zmk/kinesis-advantage360-zmk/zephyr/include/zephyr/toolchain.h:50,
                 from /__w/kinesis-advantage360-zmk/kinesis-advantage360-zmk/zephyr/include/zephyr/sys/errno_private.h:10,
                 from /__w/kinesis-advantage360-zmk/kinesis-advantage360-zmk/zephyr/lib/libc/minimal/include/errno.h:32,
                 from /__w/kinesis-advantage360-zmk/kinesis-advantage360-zmk/zephyr/include/zephyr/drivers/uart.h:23,
                 from /__w/kinesis-advantage360-zmk/kinesis-advantage360-zmk/zephyr/drivers/serial/uart_nrfx_uarte.c:11:
/__w/kinesis-advantage360-zmk/kinesis-advantage360-zmk/zephyr/soc/arm/nordic_nrf/common/./soc_nrf_common.h:218:9: error: static assertion failed: "/soc/uart@40002000 defined without required pin configuration"
  218 |         BUILD_ASSERT((IS_ENABLED(CONFIG_PINCTRL) &&                     \
      |         ^~~~~~~~~~~~
/__w/kinesis-advantage360-zmk/kinesis-advantage360-zmk/zephyr/drivers/serial/uart_nrfx_uarte.c:2078:9: note: in expansion of macro 'NRF_DT_CHECK_PIN_ASSIGNMENTS'
 2078 |         NRF_DT_CHECK_PIN_ASSIGNMENTS(UARTE(idx), 1,                            \
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/__w/kinesis-advantage360-zmk/kinesis-advantage360-zmk/zephyr/drivers/serial/uart_nrfx_uarte.c:2183:1: note: in expansion of macro 'UART_NRF_UARTE_DEVICE'
 2183 | UART_NRF_UARTE_DEVICE(0);
      | ^~~~~~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
[285/352] Linking C static library zephyr/drivers/usb/device/libdrivers__usb__device.a
[286/352] Building C object zephyr/drivers/entropy/CMakeFiles/drivers__entropy.dir/entropy_bt_hci.c.obj
[287/352] Building C object zephyr/drivers/entropy/CMakeFiles/drivers__entropy.dir/entropy_nrf5.c.obj
[288/352] Building C object zephyr/drivers/timer/CMakeFiles/drivers__timer.dir/sys_clock_init.c.obj
[289/352] Building C object zephyr/drivers/kscan/CMakeFiles/drivers__kscan.dir/__w/kinesis-advantage360-zmk/kinesis-advantage360-zmk/zmk/app/module/drivers/kscan/kscan_gpio_matrix.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build /__w/kinesis-advantage360-zmk/kinesis-advantage360-zmk/build/left
Error: Process completed with exit code 1.
```

For context, I'm not familiar at all with the internals I've tinkered with, so any help would be appreciated. Thank you in advance.

ReFil, 2023-12-27:
Hiya, you're on the right track but you shouldn't have the uart0 node there at all. Please see the official docs on how to add usb logging to a board 

https://zmk.dev/docs/development/usb-logging

Basically you only need `CONFIG_ZMK_USB_LOGGING=y`, the &usbd section and then you add the zephyr console at the chosen node

alexandregressier, 2023-12-27:
Thank you! That did the trick. Here are my exact changes for future reference:

`config/boards/arm/adv360/adv360.dtsi`

```diff
         zmk,battery = &vbatt;
         zmk,matrix_transform = &default_transform;
         zmk,underglow = &led_strip;
+        zephyr,console = &cdc_acm_uart;
     };
 
     default_transform: keymap_transform_0 {
```

```diff 
 &usbd {
     status = "okay";
+    cdc_acm_uart: cdc_acm_uart {
+        compatible = "zephyr,cdc-acm-uart";
+        label = "CDC_ACM_0";
+    };
 };
```
 
`config/boards/arm/adv360/adv360_left_defconfig`
```diff
 CONFIG_ZMK_HID_KEYBOARD_EXTENDED_REPORT=n #Set this to y for F13-F24 and intl keys with NKRO
 CONFIG_ZMK_HID_CONSUMER_REPORT_USAGES_BASIC=y
 CONFIG_BUILD_OUTPUT_UF2=y
+
+# Turn on logging, and set ZMK logging to debug output
+CONFIG_ZMK_USB_LOGGING=y
```

`config/boards/arm/adv360/adv360_right_defconfig`
```diff
 CONFIG_ZMK_HID_REPORT_TYPE_NKRO=y
 CONFIG_ZMK_HID_CONSUMER_REPORT_USAGES_BASIC=y
 CONFIG_BUILD_OUTPUT_UF2=y
+
+# Turn on logging, and set ZMK logging to debug output
+CONFIG_ZMK_USB_LOGGING=y
```

closed 2023-12-27 by alexandregressier