KinesisCorporation/Adv360-Pro-ZMK #307: How to distinguish between both volumes for flashing

alexandregressier, 2023-12-27:
Hello there,

I am willing to build an automation that will deposit the left and right firmware files whenever their respective volumes are mounted. The problem is, I cannot figure out a way to distinguish between the two volumes. Indeed, they both have the same name (`ADV360PRO` by default) and same volume UUID. I haven't tried renaming one of the volumes, but if possible I'd like to avoid doing that.

Here's what I've tried so far:

1. Looking up the volume UUID via system utilities (in my case, macOS):
```bash
diskutil info $(mount | awk '/ADV360PRO/ {print $1}') | awk '/Volume UUID/ {print $3}'
```
This prints the same UUID for both volumes.

2.  Trying to map the USB product ID and vendor ID to a mounted volume:
It seems this isn't possible.

3.  Attempting to parse info from the hex representation of the `CURRENT.uf2` firmware file:
This approach is not practical and gave up even after trying the `uf2tool` from Microsoft.

Any help would be appreciated. Thank you in advance.

ReFil, 2023-12-27:
The boards can be differentiated outside of bootloader mode but once in bootloader mode it's not possible using the stock bootloader. A 360 pro user came up with a neat script to differentiate outside of bootloader mode

https://github.com/KinesisCorporation/Adv360-Pro-ZMK/discussions/119#discussioncomment-5383832

alexandregressier, 2023-12-28:
Unfortunately, the script you linked is gone. So I've tried to go the hex parsing route again, but turns out that both firmware files contain the same exact UF2 32 bytes header.

On macOS, I've discovered that a given USB peripheral has a volume mounted if it has a "Media" key in the command `system_profiler SPUSBDataType`.
With a bit of `jq` - for which I'm by no means an expert - here's what I've been able to script out:

```bash
system_profiler SPUSBDataType -json | jq '
  any(.SPUSBDataType | .. | objects | with_entries(select(.key == "_name")); ._name == "Adv360 Pro") as $left |
  any(.SPUSBDataType | .. | objects | with_entries(select(.key == "_name")); ._name == "Adv360 Pro rt") as $right |
  any(.SPUSBDataType | .. | objects | with_entries(select(.key == "_name" or .key == "Media")); ._name == "ADV360PRO" and (.Media[].bsd_name | length) > 0) as $bootloaded |

  if $right and $bootloaded then "left"
  elif $left and $bootloaded then "right"
  else "none"
  end
'
```
Given that:
- When bootloader mode is not active, left side is listed as `Adv360 Pro` and right side as `Adv360 Pro rt`.
- If bootloader mode becomes active for either left or right side, it becomes listed as `ADV360PRO`.

This command outputs:
- `left` if right side is listed and the other side (= left) is in bootloader mode.
- `right` if left is listed and the other side (= right) is in bootloader mode.
- `none` otherwise.

closed 2023-12-28 by alexandregressier