KinesisCorporation/Adv360-Pro-ZMK #312: "Try connecting you device again" on Windows 10/11 with firmware 3

carlbolduc, 2024-01-04:
I recompiled the firmware to change the keymap on my 360 pro. I followed the instructions in the manual. The keyboard works fine when I plug it in my pc, I can see also that the mapping was successfully changed.

However, I can no longer connect my keyboard to my PC using bluetooth. I tried clearing all the profiles on the keyboard and I deleted the device from my Windows 10 PC (where it was paired before the firmware upgrade). I tried to connect it to another PC (Windows 11) where the keyboard had never been connected before. On both PC, I get this error after a few seconds of the pairing attempt:
Try connecting you device again

I tried to connect the keyboard to my iPhone and it worked.

What can I do to make my keyboard work with Windows again?

ReFil, 2024-01-04:
Have you ensured you're using a different bluetooth profile for each device?

carlbolduc, 2024-01-04:
> Have you ensured you're using a different bluetooth profile for each device?

Yes, and I tried several profiles on Windows, they all failed to connect.

ReFil, 2024-01-04:
The board will reject the connection if the host device was paired to another profile. It might be worth clearing every profile on the device even if the light is fast flashing

carlbolduc, 2024-01-04:
I cleared all the profiles and tried again, same behaviour. Have you guys tested Windows with the latest firmware?

ReFil, 2024-01-04:
Yes, every update is tested on a windows, macOS and Linux computer. 

Please try performing a settings reset using the guide linked here: https://kinesis-ergo.com/wp-content/uploads/Advantage360-Professional-Settings-Reset-Instructions-11.22.23-KB360-PRO-GBR.pdf

carlbolduc, 2024-01-06:
Thank you for the procedure. I did the reset and then proceeded to upload my firmware. Same issue as before, Windows attempts to connect and the message "Try connecting you device again" appears after a few seconds. They light continues flashing rapidly on the keyboard until it sleeps.

ReFil, 2024-01-07:
Does your computer have an intel wireless card? This could cause the problem
We've worked on a new beta firmware with changes to improve device compatibility, please find it attached.
[360pro-beta-1-6-24.zip](https://github.com/KinesisCorporation/Adv360-Pro-ZMK/files/13854699/360pro-beta-1-6-24.zip)

If you'd be able to test this and provide feedback, it would be much appreciated

carlbolduc, 2024-01-07:
Hi, both of my PCs have Intel bluetooth chipsets. I tried the firmware you provided and it worked immediately. Thank you.

Is there any way I can customize this beta firmware? My key mapping is a bit different than the stock version.

ReFil, 2024-01-07:
You can solve this problem for the time being by putting these lines into your _defconfig files
https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/278#issuecomment-1858876955

carlbolduc, 2024-01-09:
I added the lines mentioned in #278 to my config and it worked. Thank you so much for helping me fix this.

KinesisCorporation, 2024-01-09:
Thanks for your patience!

closed 2024-01-09 by KinesisCorporation