KinesisCorporation/Adv360-Pro-ZMK #321: Repeating Key with BT and USB V3.0

rpisarew, 2024-01-11:
Firmware Version: V3.0
OS: MacOs Sonoma 14.2.1

I had some issues with repeating keys in the older versions, but since V3.0 only the 'D' key is repeating even with a USB cable connected.

ReFil, 2024-01-11:
Does the problem go away if you downgrade the firmware?

rpisarew, 2024-01-11:
No it get's even worse with the older versions.

ReFil, 2024-01-11:
This doesn't sound like a software issue, please try the latest beta firmware which can be found here: https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/259#issuecomment-1880166816

If this doesnt solve the issue you might need to open a ticket: https://github.com/KinesisCorporation/Adv360-Pro-ZMK?tab=readme-ov-file#other-support

closed 2024-01-12 by rpisarew