KinesisCorporation/Adv360-Pro-ZMK #322: Upgrade instructions aren't clear to me

P05TPWND, 2024-01-11:
I’m reading all of the information on upgrading my repo and keyboard to use the latest V3.0 branch & firmware (ZMK). 
These are the resources I’m looking at:
https://github.com/KinesisCorporation/Adv360-Pro-ZMK#upgrading-from-v2-to-v3
https://github.com/KinesisCorporation/Adv360-Pro-ZMK/blob/V3.0/UPGRADE.md
and
https://kinesis-ergo.com/support/kb360pro/#firmware-updates

After reading all of these, I’m uncertain of how I move forward to upgrading. Is there more concise documentation? Some things I’m uncertain of:
1. My forked repo: I can change the default branch to V3.0, but I’m not seeing merge conflicts as described in the documentation. Is there more to it than just setting the default branch to V3.0 in my forked repo? Even testing some changes from the GUI, builds fail. 
2. Once my repo is sorted, I’m unclear if there’s some process that I must perform on my keyboard itself (i.e., resets with V2.0 and/or V3.0, etc) prior to flashing my new firmware with my custom layout.

Any help you can provide would be greatly appreciated!

ReFil, 2024-01-11:
Looking at your repository it looks like you are running into a build error due to multiple macros called `&macro_parens`. Delete these lines from your macros.dtsi file https://github.com/P05TPWND/Adv360-Pro-ZMK-NP/blob/V3.0/config/macros.dtsi#L49-L66 and it should build without problems. Because you fulfilled the full v2.0 update cycle you are not running into the merge conflict that can happen.

Try flashing the board as is and seeing if both halves connect, if they do not connect then you need to undertake the reset procedure

P05TPWND, 2024-01-11:
@ReFil Thank you for pointing this out! I completely missed the [specific error](https://github.com/P05TPWND/Adv360-Pro-ZMK-NP/actions/runs/7482849676/job/20390751216#step:9:31) regarding the macros.dsi and this was what hung me up thinking I needed to do something else regarding the repo.

closed 2024-01-11 by P05TPWND