KinesisCorporation/Adv360-Pro-ZMK #347: Problem with Kinesis Advanced 360 Pro

Manubi, 2024-01-19:
I wanted to migrate from V2.0 to V3.0 of the firmware.

I run on an apple macbook pro intel 2019 macOS Sonoma 14.2.1

**Problem 1**: when i put the keyboard in bootloader mode (3 green lights) and i just drag the .uf2 file from the finder onto the volume ADV360PRO i get the following error:
[image]

After a while of googling I was able to fix it with simply copying with 'cp left.uf2 /Volumes/ADV360PRO' which did work till now.

**Problem 2:** Now if I want to update the keyboard in the terminal with the cp command I get the following error:

[image]

The keyboard goes into the copy mode... blinks gives me the above error and jumps back into bootloader mode (3 green lights).

The only reason I wanted to try to update to firmware V3 was to try again if the bluetooth would work with my mac...

Has anyone an idea how i an fix this mess?

Thanks

ReFil, 2024-01-19:
The board jumping back into bootloader mode after flashing the settings reset file is expected behaviour.

The finder error is not indicative of problems. Please see this note from ZMK on the topic: https://zmk.dev/docs/troubleshooting#file-transfer-error

Manubi, 2024-01-19:
I downgraded back to V2.0 and now finally everything is working again including the bluetooth connection. I guess there is no real benefit in updating if everything is working fine?

ReFil, 2024-01-19:
V3.0 brings additional features over V2.0 but it's you're happy with it as is you don't need to upgrade

ReFil, 2024-01-23:
Closing issue as solved

closed 2024-01-23 by ReFil