KinesisCorporation/Adv360-Pro-ZMK #364: V2.0 version

GGuinea, 2024-01-31:
Hi!

After updating to V3.0 I'm struggling with communication problems between keyboard and newest macbook pro Sonoma 14.2.1. I want to rollback to old v2.0 version where everything was fine. Where Can I find old tag?

ReFil, 2024-01-31:
I would recommend you try this change: https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/104#issuecomment-1896681556

huber-th, 2024-01-31:
I had this issue as well where connection would randomly drop to my MacBook running latest Sonoma.

No drops since changing `CONFIG_ZMK_BLE_EXPERIMENTAL_FEATURES=y` to `CONFIG_ZMK_BLE_EXPERIMENTAL_FEATURES=n` in `adv360_left_defconfig` and `adv360_right_defconfig`

asilvam133, 2024-01-31:
Hi @ReFil. I applied the change you recommended and it worked.
Will this be fixed?

I notice that since the upgrade my keyboard connects to my Macbook as soon as it turns on, so that is a huge win and I was wondering if that new behavior is part of that feature being enabled?

ReFil, 2024-01-31:
This is a known bug currently. The fix for #278 can lead to connectivity issues on macOS. This will be addressed in a future update

ReFil, 2024-02-06:
Just following up on this, did those configuration changes rectify the issues you're having?

GGuinea, 2024-02-06:
I've tested it extensively and it helped. 
@ReFil should I close this issue? Maybe it should be open since it's not only my problem and probably it would be easier to find fix for other users

kbumsik, 2024-03-01:
I have two Adv360 keyboards and both had exactly the same issue after upgrading.

I also confirm that setting `CONFIG_ZMK_BLE_EXPERIMENTAL_FEATURES=y` fixes the issue on both keyboards!

ztomer, 2024-03-13:
I can confirm to also experience this issue on an M1 Macbook Pro with Sonoma 14.4. 
With CONFIG_ZMK_BLE_EXPERIMENTAL_FEATURES=y

ReFil, 2024-04-15:
This should now be rectified with the latest update

closed 2024-04-15 by ReFil

kbumsik, 2024-04-16:
I applied the new update to two Adv360s for a day right after the release.
I report that I have no such issue for a day!