KinesisCorporation/Adv360-Pro-ZMK #368: Adv360 Randomnly Disconnects & Reconnects

nickcoffey, 2024-02-02:
# The Problem
I've been having trouble with my Adv360 Pro randomly disconnecting from bluetooth while typing. It usually reconnects in a second or two. Sometimes when this happens the computer will keep receiving the last key pressed until it reconnects.                                                                                                    This has put a serious damper on my workflow as it is happening quite frequently.

# Attempted Solutions
I have followed the troubleshooting tips by applying the `settings-reset.uf2` to each side followed by applying my `left.uf2` & `right.uf2` and still no luck. Before starting this process I made sure my fork was up to date with the base repo.

# Other Helpful Info
This is happening on 2 separate Macbooks. One's a Macbook Pro M1 Pro and the other is a Macbook Air M2. They are both running MacOS Sonoma v14.2.1.

The link to my personal fork is [here](https://github.com/nickcoffey/Adv360-Pro-ZMK).

nickcoffey, 2024-02-02:
After double checking the other issues, I see there may be a temporary fix mentioned in #364. I'll see if flipping `CONFIG_ZMK_BLE_EXPERIMENTAL_FEATURES=y` to `CONFIG_ZMK_BLE_EXPERIMENTAL_FEATURES=n` will also solve my problems. If so, I'll report back and close this issue.

ReFil, 2024-02-06:
Did those config changes solve the problem?

nickcoffey, 2024-02-06:
Yes @ReFil they did! Thanks for checking in. I wanted to test the change for a few work days to make sure it completely solved the problem. I'm closing the issue now.

closed 2024-02-06 by nickcoffey