KinesisCorporation/Adv360-Pro-ZMK #373: Using a macro with a backtick (`) in its name causes CI to fail

gridlocdev, 2024-02-08:
In the web-based editor (https://kinesiscorporation.github.io/Adv360-Pro-GUI/), if you add a macro with a back-tick character ` in its name and commit changes, the CI build fails.

[image]

Here are a couple of my workflow runs that reproduce the issue:
- Without macro (Passed CI):  https://github.com/gridlocdev/Adv360-Pro-ZMK/actions/runs/7824034749  
- With macro (Failed CI): https://github.com/gridlocdev/Adv360-Pro-ZMK/actions/runs/7824079947

gridlocdev, 2024-02-08:
> It's my understanding backticks are not a permitted character to use for labels in the devicetree syntax used by the keymap files

If it isn't part of the ZMK spec, would it be possible to add a JavaScript input validation handler on keypress for the macro name textbox to prevent invalid characters from being added? 

[image]

(Looking at some of the DOM nodes it seems as though the editor uses Vue.js, so the documentation for implementing something like that might be found here: https://vuejs.org/guide/essentials/event-handling.html)

I believe this change would also probably be in whichever private repository publishes the client bundle to https://github.com/KinesisCorporation/Adv360-Pro-GUI.

ReFil, 2024-02-12:
I will pass this feedback onto the developer of the web GUI, as that's not my area. Thanks for bringing it to my attention

closed 2024-02-12 by ReFil