KinesisCorporation/Adv360-Pro-ZMK #377: How to use &hm (homerow mods)

RudolfVonKrugstein, 2024-02-12:
I can see that a homerow mod behavior is defined here: https://github.com/KinesisCorporation/Adv360-Pro-ZMK/blob/V3.0/config/adv360.keymap#L20

But i dont get how I can use that in the editor. I try to use &hm but It says that it does not know it and also does not give me the option of specifying mod and key.

Thanks

ReFil, 2024-02-12:
the &hm behaviour is a predefined variant of the hold-tap functionality provided by ZMK https://zmk.dev/docs/behaviors/hold-tap . It's an advanced function and as such can't be configured in the Kinesis GUI. You can either use a different ZMK GUI editor (I believe this one supports hold tap configuration: https://nickcoutsos.github.io/keymap-editor/) or do it manually in the file

closed 2024-02-12 by ReFil

RudolfVonKrugstein, 2024-02-13:
thank you, manuell editing worked fine :)