KinesisCorporation/Adv360-Pro-ZMK #380: (Feature) Add output selection to default mod layer to prefer BLE over USB.

huber-th, 2024-02-15:
By default, output (key press etc.) is sent to USB when both USB and BLE are connected. This means it is not possible to use the keyboard with a different device than the one charging the keyboard. In my personal use case I have a stationary system which I use to charge both sides of the keyboard with and to which I have the provided USB cables connected. However, while it is charging I would like to still be able to use the keyboard for other devices (e.g. another laptop, tablet or phone).

The output selection behaviour in ZMK provides support for this use case by switching ZMK to prefer BLE over USB. I suggest including two key bindings `&out OUT_BLE` and `&out OUT_USB` into the default mod layer to set the preferred output to either USB and BLE and add documentation to the readme on how to use these key mappings to change the keyboards behaviour when connected via USB as desired.

This setting will be saved to flash storage and hence persist across restarts and firmware flashes and does not constantly be selected.

The keyboard is still being detected as USB drive, when in boot loader mode for flashing, by the system it is connected via USB. No need to change the output selection prior to flashing new firmware.

More information can be found here: https://zmk.dev/docs/behaviors/outputs

If desired I would be willing to prepare the change and open a pull request.

ReFil, 2024-02-16:
Whilst I understand the request for this feature changes to the stock keymap are typically not something we want to merge unless for a very good reason because they have the potential to cause massive merge conflicts for forks. It could definitely be better documented though, I agree

huber-th, 2024-02-17:
That's a fair point. It most probably would create merge conflicts for most people if it were added to the default keymap. 

Maybe something to include in a future update to the manual/website. Feel free to close this issue.

closed 2024-02-17 by huber-th