KinesisCorporation/Adv360-Pro-ZMK #383: Where is the release commit?

toddjonker, 2024-02-16:
I've updated my firmware using the publicly-release snapshot https://kinesis-ergo.com/download/adv360-pro-factory-default-branch-3-0-v1-16-24-commit-ddcdb88/

I now need to manually remap everything to support my macOS/Dvorak layout, and I want to do so atop that same curated commit.

However, the commit ddcdb88 is not in this repository. Nor is there a single branch or tag in this repository for anything other than the development tip.

Please make a `release` branch available that tracks the canonical public releases, so that users can fork/track those and not a development branch.

ReFil, 2024-02-16:
The factory firmware tracks the firmware here, but has an additional commit to change the keymap. When changes are made here that change the actual firmware (e.g. not changes to the makefile or ci processes) the factory firmware gets updated. 

The branch here can be considered the authoritative release branch, as we have another mechanism for development/beta testing that can be seen in the readme

toddjonker, 2024-02-17:
Thanks for the prompt reply.  I feel like I've seen a lot of churn on the GitHub issues so I have been under the impression that this repository is not particularly stable.  Thus I've held off on the 3.0 update until forced into it today when my right keywell suddenly refused to connect.

I'm also curious why there's keycap changes to the factory firmware that aren't in-sync with the baseline you provide to customers that need to customize anything.  (Which AFAICT includes everyone who's not on a Windows+QWERTY layout.)

ReFil, 2024-02-17:
The factory firmware is the keymap here, except with the four macro keys mapped to keycodes to facilitate testing during assembly. It is expected that customers who will be changing the default keymap will have their own uses for these four keys. There are a few different default firmwares available to download here including Dvorak: https://kinesis-ergo.com/support/kb360pro

I can understand the concern about stability, but we're always trying to improve the experience with each update

ReFil, 2024-03-01:
I think i'm going to close this issue as solved now, for the avoidance of doubt we always recommend using the latest state of this repo to base any keymap customisations off of, and fetching upstream when new commits are added

closed 2024-03-01 by ReFil