KinesisCorporation/Adv360-Pro-ZMK #397: keymap files used to layout Dvorak Branch 3.0 (v1-22-24)

hh, 2024-03-01:
It seems the default factory image is here: 
https://kinesis-ergo.com/download/adv360-pro-factory-default-branch-3-0-v1-16-24-commit-ddcdb88/

And the factory Dvorak image is here:
https://kinesis-ergo.com/download/adv360-pro-dvorak/

I'd like to have a branch or commit that includes the Pre-Programmed Dvorak layout.
This would make an easier starting point for customization.

KinesisCorporation, 2024-03-01:
Thanks for the suggestion, this is under evaluation. The primary concern is it could make it more complicated/cumbersome for all the non-Dvorak users, but if there is an easy way to do it and support it on an ongoing basis that could make sense.

huber-th, 2024-03-02:
@KinesisCorporation an easy way to achieve this would be to include other keyboard layouts in the `config` folder of the repository in a separate keymap file. e.g. `dvorak.keymap`, `colemak.keymap`, `colemak-dh.keymap` etc. while keeping the default qwerty in `adv360.keymap` to avoid issues with existing forks.

For example if you added your Dvorak keymap as `dvorak.keymap` then all users who wish to use that layout would have to do in their own repositories is update `adv360_left.keymap` and `adv360_right.keymap` to include the `dvorak.keymap` instead of `adv360.keymap`.

```
/*
*
* Copyright (c) 2021 Polarity Works
* SPDX-License-Identifier: MIT
*
*/

#include "dvorak.keymap"
```

This would allow you to publish alternative keymap layouts without breaking current user's repositories and at the same time offer alternative layouts for users with minimal effort to switch between them.

I have done this in my own [repository](https://github.com/huber-th/zmk-adv360-pro/tree/personal/config) to test and verify by moving my customized keymap into a `personal.keymap` file and adjusted the includes in `adv360_left.keymap` and `adv360_right.keymap`.

ReFil, 2024-03-02:
It's worth noting doing so would break the online keymap configurator

hh, 2024-03-02:
Might be useful to update the key so editor to choose from a list of keymap
files to edit.
adv360, Mac, windows, and Dvorak among a few of the options.

huber-th, 2024-03-02:
> It's worth noting doing so would break the online keymap configurator

That's a good point to note. That seems more an issue of making sure there are clear instructions of how to use alternative layouts provided in a separate file. 

Instead of updating the `adv360_left.keymap` and `adv360_right.keymap` users who wish to use the online editor can rename or delete the stock `adv360.keymap` and rename the alternative layout keymap file to `adv360.keymap`?

Or simply copy the full file content from one to the other file.

abreinig, 2024-06-17:
In my opinion a good way to accomplish this and benefit everyone is to have presets that you can load in the online config editor. It could be a button or a drop down with confirmation or similar. I just got a board and had to create my dvorak layout from scratch and I was surprised that I had to do it that way.