KinesisCorporation/Adv360-Pro-ZMK #401: Wrong keys generated by PIPE and PIPE2

sedrik, 2024-03-07:
Hi

Looking at https://zmk.dev/docs/codes/keyboard-keypad#symbols--punctuation there is both PIPE and PIPE2.

I have tried to set these on my keyboard but PIPE produces and asterisk (*) and PIPE2 produces a greater than symbol (>)

As it stands right now I am missing the Scandinavian key for less than, greater than (shift) and pipe (alt gr) on my keyboard. How can I get access to that key?

ReFil, 2024-03-07:
Those keycodes are probably being altered by the layout setting on your host OS

sedrik, 2024-03-07:
Any way to confirm that assumption? I'm running a pretty standard Ubuntu install with no changes to the default layout settings.

ReFil, 2024-03-07:
In keyboard settings on your computer the layout should be set to "English (US)" for the keycodes in ZMK to line up with the expected output, as all HID keycodes were defined with that in mind. If you want to create a layout for alternative country layouts you could try using https://github.com/joelspadin/zmk-locale-generator (Although i must admit I have no experience with this tool)

sedrik, 2024-03-07:
Ah thanks that make some of it work (lost my Swedish letters now but will have to look into fixing that)

closed 2024-03-07 by sedrik