KinesisCorporation/Adv360-Pro-ZMK #410: NOT ENABLE INT keys at Mac OS Sonoma

yuuna, 2024-03-16:
Helllo

I set adv360_left_defconfig
`CONFIG_ZMK_HID_KEYBOARD_EXTENDED_REPORT=y `

but I can't enable INT keys(INT1..5)

so I check BT logs at my Mac, I found that only after I enable this config , it causes R2CAP errors.

[image]

OS: Mac Sonoma 14.3.1
Hardware: Mac Studio 2022(M1 Ultra)

ReFil, 2024-03-16:
After changing that setting you need to unpair and re pair the keyboard to the computer

yuuna, 2024-03-16:
@ReFil 

I unpair the keyboard ( Mod + Winkey) and delete profile from Mac and reboot it.
But it still show same error.

ReFil, 2024-03-16:
Is N key rollover enabled?

yuuna, 2024-03-16:
adv360_left_defconfig
```
CONFIG_ZMK_HID_REPORT_TYPE_NKRO =y
CONFIG_ZMK_HID_KEYBOARD_EXTENDED_REPORT=y
```

After I set NKRO=n , build if failed.
[image]

yuuna, 2024-03-16:
I think parameter will changed CONFIG_ZMK_HID_KEYBOARD_EXTENDED_REPORT to CONFIG_ZMK_HID_KEYBOARD_NKRO_EXTENDED_REPORT.

I will change this parameter.

https://zmk.dev/docs/config/system

ReFil, 2024-03-25:
Hi, Did changing that updated parameter work for you?

ReFil, 2024-04-15:
hello, just following up on this, is it working?

ReFil, 2024-04-21:
Hello, just following up again, if i dont hear from you in the next 24 hours i will close this issue as solved, thank you

closed 2024-04-24 by ReFil