KinesisCorporation/Adv360-Pro-ZMK #427: Possible typo in `adv360_left_defconfig`?

khoek, 2024-04-05:
I can't seem to find the config var `CONFIG_ZMK_HID_KEYBOARD_EXTENDED_REPORT` referenced in `adv360_left_defconfig` in the main ZMK repo, but I can find the very similar `CONFIG_ZMK_HID_KEYBOARD_NKRO_EXTENDED_REPORT` (noticed while discovering https://github.com/zmkfirmware/zmk/issues/2253 actually). Is this a typo/old name from when the option was being developed?

Confusingly, this other name seemed to me to work just as well as the documented one, but I might have been too hasty in my testing. Perhaps this is all fine and there is some sort of compatibility/alias that I've missed which explains this?

khoek, 2024-04-05:
Ah! I see it's in the custom fork. No worries.

closed 2024-04-05 by khoek

ReFil, 2024-04-05:
This feature was developed and implemented in the custom repo first, then upstreamed later. During the upstreaming process the zmk maintainers wanted the name of the kconfig option changed to better describe the option. As a result we now have an additional kconfig option  in the custom fork to assign it properly for users