KinesisCorporation/Adv360-Pro-ZMK #438: Keyboard halves don't connect to each other after PC wakes up from sleep

totkeks, 2024-04-12:
I am using a Windows 11 PC and the Advantage 360 Pro with the 3.0 firmware.

I have the issue that the halves don't connect to each other after my PC wakes up from sleep. I only have the three LEDs blinking in red when pressing any key.

From my observations it only happens when the keyboard is connected to the PC via an USB hub that is powered down when the PC goes to sleep, e.g. the USB hub in my monitor or my new powered USB hub (it doesn't use/receive power from the PC, neither sleeping nor awake).

In contrast, when I connected the keyboard halves to my PC directly this error never happened.

Is there any way to increase the delay for when the two halves try to find each other? Or another way to fix this besides unplugging the cables and plugging them again?

ReFil, 2024-04-15:
is there any activity from the left side when this happens i.e. layer light lighting up when you push the kp layer key?

totkeks, 2024-04-16:
> is there any activity from the left side when this happens i.e. layer light lighting up when you push the kp layer key?

Luckily, I just had it happen again.

No activity on the left side when I click the kp or fn keys.
I also just noticed that the red blinking LEDs are only present on the right side, not on the left one.

Does this mean the left side didn't power on / boot properly?

ReFil, 2024-04-16:
yes, are you running the latest available firmware? it's possible you're encountering a bug in older versions of the firmware

totkeks, 2024-04-16:
I was not, but I just updated [my repo](https://github.com/totkeks/Advantage360Pro-Firmware) locally with the latest changes from the `V3.0` branch and built and flashed the new firmware.
Sent my PC to sleep, woke it up again and had the same issue again.

Is there any way I could debug this or retrieve a log file from the left side?

Btw, the backlight was on after waking the PC, so at least it was powered. If that is any help.

ReFil, 2024-04-16:
Neither of your branches on that repo look to be up to date. Can you try the latest unchanged firmware first please (from here: https://github.com/KinesisCorporation/Adv360-Pro-ZMK/actions/runs/8709784894) and see if that works?

totkeks, 2024-04-16:
That is right. I just did a quick merge locally without pushing it.

I just flashed the version you linked, sent the PC to sleep and woke it up. Same error as before.

ReFil, 2024-04-16:
It sounds like there's some weird power management conflict going on. You can enable UART logging but it uses the usb connection for it. I haven't been able to replicate this crash in my testing. Maybe turning off some of the power management settings for the usb hub in device manager will help?

NullFragment, 2024-05-16:
I think this has more to do with the order in which both halves of the keyboard regain power if you have them plugged in and the power switches set to the off position. I've noticed that if I plug my keyboard halves into a hub vs the computer itself, sometimes it will be fine and other times I'll have to unplug and replug both halves of the keyboard back in.

That said, it only happens if I don't have the power for the halves switched on so they maintain power during the brief periods in which the hub doesn't supply power to them.

KinesisCorporation, 2024-05-17:
The power-on sequence was **definitely** a factor in 1.0 firmware (syncing worked best when the left module had a moment to initialize first, before turning on the right) but hasn't really been an issue for a while to our knowledge.

@NullFragment It looks like you've got a 3.0 build so you should **not** be seeing this unless there is something unique about the Hub. Are you able to reproduce failure-to-sync-on-startup just by turning on the battery on the right module, then the left.

NullFragment, 2024-05-17:
Oddly enough, I haven't been able to reproduce it since I wrote that comment. It was only happening when using both modules wired with the power turned off. The power sequence didn't seem to matter if I was using wireless, but I've tried both unplugging the keyboards from the USB hubs as well as the hub itself from the PC and couldn't repro it again.

duanemay, 2024-05-18:
I recently changed my setup where my keyboard plugs into my monitor's hub and switches between two computers. After the monitor goes to sleep, it cuts power to the USB devices, and after wake up I also get the blinking red lights on the right half of the keyboard. I am on the latest 3.0 firmware. 

As a work around, I changed settings on the monitor to keep the USB powered on during sleep. However, this also makes it so the monitor does not always go to sleep.

KinesisCorporation, 2024-05-20:
Hi Duane,

That's also surprising behavior. Can you run the "Version Macro" command (Mod + V) for us here so we can see what firmware you are runng

bartlomiej-mucha-bpms, 2024-07-29:
Hi I have similar issue since a long time.
I wanted to create an issue for that but I see there is a similar thread here.

My customization branch on top of V3. https://github.com/bartlomiejmucha/Adv360-Pro-ZMK/tree/my-customisation-v3
I don't use bluetooth so I disabled it. 
I connect the keyboard to the Caldigit Elements dock together with Kensington Expert Mouse Trackball, usb-c to DisplayPort dongle for my monitor and usb-c enthernet dongle. The dock itself is connected to the MacBook Air with the latest OS.
Many times during a day I have to unplug MacBook and take it to other place, when I get back I plug it in to the dock, and all devices works fine except the keyboard. 
So I have to unplug the keyboard (both halves) and then plug them in starting wth the left one, wait a sec, then the right one like in the issue https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/510

That is a bit annoying as I have to do it many times everyday. I worry that I will break the usb ports on the keyboard.

bartlomiejmucha, 2024-07-29:
Hi I have similar issue since a long time.
I wanted to create an issue for that but I see there is a similar thread here.

My customization branch on top of V3. https://github.com/bartlomiejmucha/Adv360-Pro-ZMK/tree/my-customisation-v3
I don't use bluetooth so I disabled it.
I connect the keyboard to the Caldigit Elements dock together with Kensington Expert Mouse Trackball, usb-c to DisplayPort dongle for my monitor and usb-c enthernet dongle. The dock itself is connected to the MacBook Air with the latest OS.
Many times during a day I have to unplug MacBook and take it to other place, when I get back I plug it in to the dock, and all devices works fine except the keyboard.
So I have to unplug the keyboard (both halves) and then plug them in starting wth the left one, wait a sec, then the right one like in the issue https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/510

That is a bit annoying as I have to do it many times everyday. I worry that I will break the usb ports on the keyboard.

totkeks, 2024-08-16:
Wanted to bring an update here. I recently, like yesterday, noticed that I don't have the issue with disconnected halves after waking the PC up from sleep anymore.

Since I didn't upgrade the firmware of my keyboard since posting here, I assume Windows did some making update for their USB drivers. I'm running Windows Dev Insider builds though - currently build 26120.1350.

totkeks, 2024-12-07:
Update after a while, it still doesn't work. Got even worse. But I think it is my weird USB hub, that I have the halves connected to for power, and not the keyboard itself.

@ReFil is there any way to debug log everything the keyboard does and then send out via BLE or USB?

ReFil, 2025-01-17:
A new update has been created for the 360 Pro that includes some fixes that should help with this situation. Precompiled firmware files are available here: https://kinesis-ergo.com/download/advantage360-pro-clique-beta-01-16-25/ for users without significantly complex custom keymaps, alternatively you can try building off the currently open pull request (#630) as well

totkeks, 2025-01-17:
> A new update has been created for the 360 Pro that includes some fixes that should help with this situation. Precompiled firmware files are available here: [...]

Wow, cool. I just read the PR and the ZMK studio and clique. Gonna test that soon.
This will certainly make exiting layouts much easier for people that are not used to git or didn't like the old web editor.

bartlomiejmucha, 2025-02-06:
Hi, looks like clique update solved the issue for me. Thanks

totkeks, 2025-02-06:
Yeah, same here. Not that you mentioned it, didn't have that issue anymore since updating to the new ZMK version

PR #630 (2025-02-06): Clique and mouse movement update
### What's changed:

This change updates the advantage 360 Pro to a newer version of ZMK, incorporating support for mouse movement behaviors, as well as the ZMK Studio / Kinesis Clique real time keymap editing system. Furthermore a number of additional bugfixes and changes have been implemented to improve the user experience

Additional information on Clique can be found here: https://kinesis-ergo.com/360p-clique-upgrade/

### Why has this change been implemented:

This adds mouse movement support into the Advantage 360 Pro as well as rectifying several outstan [... cut]
closed 2025-02-06 by totkeks