KinesisCorporation/Adv360-Pro-ZMK #450: Keyboard not detected after hard reset

abbycross, 2024-04-17:
Hi! Today I unplugged my Kineseis Advantage2 (KB600), then when I plugged back in a bit later, the keyboard would not respond. When I do a hard reset, I do see the four blinking LEDs, but it still does not work other than those lights. My laptop (Mac Powerbook) does not detect the keyboard. Any advice would be much appreciated. Thank you!!

KinesisCorporation, 2024-04-17:
Hi Abby, This forum is for questions pertaining exclusively to the newer Advantage360 keyboard which has a totally different firmware engine than the Advantage2 and a different set of engineers. Sorry if that wasn't super clear.

We are sorry to hear you are having trouble with your Advantage2. Please shoot us a quick ticket and we'll get the troubleshooting process started with you over email. 

https://kinesis-ergo.com/support/contact-a-technician/

Have a great day!

abbycross, 2024-04-17:
Whoops sorry I didn't read carefully!