KinesisCorporation/Adv360-Pro-ZMK #453: Receive error code -36 after when trying to flash the firmware

ujwal-jain, 2024-04-20:
Hello, I follow instructions to flash the firmware after modifying   my layout with the GUI for the Advantage Kinesis 360 pro. When i try to copy the [uuid]-left.uf2 to my left keyboard or right keyboard in bootloader mode, the program crashes and i get the error -36. 
[image]

I'm struggling to figure out if it's something I've missed or if it's a faulty device. Appreciate any help, thanks!

ReFil, 2024-04-20:
Have you used the mod+v key combination to check the version flashed to the board? Finder is known to throw error codes but the board still flashes

ReFil, 2024-04-30:
Hello, just following up on this, has the firmware been properly updated?

ReFil, 2024-05-12:
Closing issue as solved, the -36 issue is known to be spurious and does not stop the board flashign

closed 2024-05-12 by ReFil