KinesisCorporation/Adv360-Pro-ZMK #459: Unable to flash 360--permissions error

kelvinfann, 2024-04-25:
```
% sudo cp firmware/202404250542-d4b2204-left.uf2 ~/../../Volumes/ADV360PRO/left.uf2
Password:
cp: /Users/kelvinfann/../../Volumes/ADV360PRO/left.uf2: Operation not permitted
```

When I try to copy the firmware files to set the layout I am not allowed to? 

Running on a mac Sonoma 14.4.1 and the build seemed fine

kamalkech, 2024-04-26:
with expensive price should support client
@kelvinfann me too i get the same error

KinesisCorporation, 2024-04-26:
Hi folks, 

We just tested copying files to a 360 Pro here from a MBP Running Sonoma 14.4.1. We got the now "standard" faux macOS error about file transfer but the update was nonetheless successful, so this doesn't seem to be a keyboard issue per se.

- Have you tried running the version Macro (Mod + V) to see if the update was in fact successful?
- If it wasn't, is there another PC you can test on?
- Can you test our factory firmware files? https://kinesis-ergo.com/support/kb360pro/#firmware-updates

@ReFil Any thoughts?

[image]

[image]

KinesisCorporation, 2024-04-29:
Were you able to get it working?

kelvinfann, 2024-04-30:
> Hi folks,
> 
> We just tested copying files to a 360 Pro here from a MBP Running Sonoma 14.4.1. We got the now "standard" faux macOS error about file transfer but the update was nonetheless successful, so this doesn't seem to be a keyboard issue per se.
> 
> * Have you tried running the version Macro (Mod + V) to see if the update was in fact successful?
> * If it wasn't, is there another PC you can test on?
> * Can you test our factory firmware files? https://kinesis-ergo.com/support/kb360pro/#firmware-updates
> 
> @ReFil Any thoughts?
> 
> <img alt="image" width="273" src="https://private-user-images.githubusercontent.com/37431414/326077153-a98c6ea1-cc60-4bf1-abfa-306c2de9a1c3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ0NDg1MjMsIm5iZiI6MTcxNDQ0ODIyMywicGF0aCI6Ii8zNzQzMTQxNC8zMjYwNzcxNTMtYTk4YzZlYTEtY2M2MC00YmYxLWFiZmEtMzA2YzJkZTlhMWMzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDMwVDAzMzcwM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTViNTI1ZTkyOGU3ODA5NjBlNDE2NmVlMWZjZWJkZWNiY2NmZDFhZjc1MjA0ZGZmOGQ0ZTk0OTRlMzJhOTUzOTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.jncCQ4MU5AkpCPC5XEc7iudOSdhUUHV-xHfEApRRnsk"> <img alt="image" width="407" src="https://private-user-images.githubusercontent.com/37431414/326076947-d90f72c1-0244-48ec-a474-085c715feb37.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ0NDg1MjMsIm5iZiI6MTcxNDQ0ODIyMywicGF0aCI6Ii8zNzQzMTQxNC8zMjYwNzY5NDctZDkwZjcyYzEtMDI0NC00OGVjLWE0NzQtMDg1YzcxNWZlYjM3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDMwVDAzMzcwM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWYzNzQ4MmQ5YThmYzhhNGRmN2IyY2U0ODdlN2QyZmZjYjhlZDRlZDc4YWU5OGE2ZTA2MGE4ZTNmYjM1NTliMWYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.DKhYF5kM03j8MpOZ4vb-00ozU8vw4HBaXTBxFWgmbRM">

1. Using the macro the version is not updated--and you can tell because the remapped 
2. I tried on a couple of macs and none seem to work unfortunately
3. I'll try the firmware tonight and get back

ReFil, 2024-04-30:
Have you tried dragging and dropping the firmware on to the board in finder? I wonder if changing the destination name is throwing the bootloader off. It doesn't work like a real USB drive, it's emulating mass storage functionality so it may not support the rename operation, hence the permission denied error

GoByeBye, 2024-04-30:
It seems the issue is that you're trying to rename the file along with copying it. 
Same issue happens on Ubuntu 24.04 LTS if you try to `cp` with a rename of the file.

@ReFil is probably onto something [in his comment](https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/459#issuecomment-2084535354)

The proper command to copy it would be `cp firmware/202404250542-d4b2204-left.uf2 ~/../../Volumes/ADV360PRO/` or `cp firmware/202404250542-d4b2204-left.uf2 ~/../../Volumes/ADV360PRO/202404250542-d4b2204-left.uf2` as that leaves the file name intact

kelvinfann, 2024-05-02:
@GoByeBye I actually ran those cp commands first before I tried anything in finder--still no dice; I think this device has to go back for me unfortunately. Back to the advantage 2 I go. I'll check back in like 3 months when I switch desktop setups to see if this issue is resolved

ReFil, 2024-05-02:
@kelvinfann will the file not copy over in finder either?

abreinig, 2024-06-21:
You can also try to flash with your mobile phone if you have Android. I'm not sure if it'll work with iPhone.

pavveu, 2024-07-12:
i have the same issue on mac, seems to be the problem with the partition schema (when checking in terminal) but not sure if running command for repairing the partition schema is good idea... so probably will skip it for now. Checked permission - seems good, checked free space, also good. 

$ diskutil verifyDisk /Volumes/ADV360PRO
Nonexistent, unknown, or damaged partition map scheme
If you are sure this disk contains a (damaged) APM, MBR, or GPT partition
scheme, you might be able to repair it with "diskutil repairDisk /Volumes/ADV360PRO"

ReFil, 2024-07-12:
Writing the partition scheme will do nothing, the board bootloader will not recognise that as a valid command

pavveu, 2024-07-12:
will try from my windows desktop, but if it doesnt work does it mean its something with the drive on the keyboard?
also i noticed that the files that i have installed are from 2022 - does it mean i need to run some upgrade before flashing ZMK (got the keyboard today, wanted to change the layout and remap some keys) 

[image]

ReFil, 2024-07-12:
The date on those files is the compile date of the bootloader, it bears no relevance to the firmware you can install. The drive is not a real drive, it is emulated, if the main firmware works properly it is highly unlikely there is anything wrong

pavveu, 2024-07-12:
i've used mod+v to print the version, it seems it was uploaded - its printing the date and version hash, so it seems to work but some of the keys are not working on the FN layer 
Basically fn+ (= and 1-9-0) works as f1 to f12 
but the lower the keys below it, doesn't. the mod3 key - where i assigned f12 works. On other keyboards that i have f13-f24 works normally. So it's not some system setting i assume, in this configuration -> as seen on the screen f13-f24 doesnt work

[image]

pavveu, 2024-07-12:
also i assigned some keys under num layer - and they work. i cannot understand why in this scenario f13-f14 and rest doesnt work as intended

KinesisCorporation, 2024-07-12:
F13-F24 were in conflict with NKRO. Trying disabling NKRO.

https://kinesis-ergo.com/support/kb360pro/#faq

pavveu, 2024-07-12:
works like charm, thanks for the patience ;)

KinesisCorporation, 2024-07-12:
Great!

closed 2024-07-12 by KinesisCorporation