KinesisCorporation/Adv360-Pro-ZMK #467: Flashing firmware from Mac

goDawgs18, 2024-05-07:
Hi,

I have:
1.  compiled the firmware 
2. Got the keyboard into bootloader mode
3. Mounted the Fat 16 drive.

However now I'm stuck because I cannot copy over the firmware into the directory because it is in read-only mode (or something). 

When I look at the info for the directory it is marked as "You can only read". So as I mentioned I am unable to cannot copy the firmware over. 

During my investigation I have even found a reddit post documenting my same issue: https://www.reddit.com/r/kinesisadvantage/comments/15osdmq/i_am_unable_to_flash_firmware_on_the_360_pro_due/

- Using Advantage 360
- Apple silicon 
- OSX: 14.3.1 (23D60)

goDawgs18, 2024-05-07:
Seems like I'm not alone: https://github.com/KinesisCorporation/Adv360-Pro-ZMK/issues/459#issue-2262751699

Probably going to have to return the device as well 😞

KinesisCorporation, 2024-05-07:
Hi there, This is not a keyboard limitation as far as we know. Is there another PC you can try mounting the drive on?

goDawgs18, 2024-05-08:
There is another Mac that I can try to use? But I'd expect the same behavior.

goDawgs18, 2024-05-08:
I guess may as well try

KinesisCorporation, 2024-05-08:
How did it go with the other computer?

goDawgs18, 2024-05-08:
I don’t have usual access to it. It’ll be about a week or so.

KinesisCorporation, 2024-05-08:
Great. If it doesn't work on the other computer we are happy to do an exchange (link below). Nut it seems unlikely (although certainly not impossible) that this is a keyboard hardware problem.

https://kinesis-ergo.com/support/contact-a-technician/

ReFil, 2024-05-20:
Just following up on this @goDawgs18 did flashing it from the other computer rectify the issue?

apg1, 2024-07-26:
Had the same issue then went to a windows machine used it, it likes to disconnect as soon as you add the files, but it worked on OSX after that.

dylan-chong, 2024-10-19:
did anyone figure out how to flash the firmware from mac?

apg1, 2024-10-19:
> did anyone figure out how to flash the firmware from mac?

Yes, I flash from mac without issue. The thing that caught me off guard was that when I flash it immediately drops the connected drive, that's because the unit (when it gets a new config) reboots itself to load it. 

To flash from Mac I connect it via USB (usually their cable but not all the time).
Turn off the power switch on the unit you are flashing.
I update the right side first, so I connect the right side then use a bent paper clip to get into bootloader mode, it shows up on my mac as a new drive. 
Open the drive, and drop the RH file in there and it will immediately disconnect. 
I press the mod key on the left unit to keep it awake while the right reboots.

I then hustle and do the same with the left side. 

That seems to work, maybe more steps then needed but it seems to work consistently.

ainformatico, 2024-11-27:
Same steps as @apg1 and has been working consistently for 3 months now.
Things to remark that also were strange for me:
* It is "normal" the error about the disk not being connected anymore after flashing. I just close the error pop-up. It is because the unit restarts. Happens for both units.
* Start with the right side. When I did the left side first, I had inconsistent errors and had to manually reset it (using the hidden reset button)
* In case you don't know it, this is the best companion to do more advance setup of your layout https://nickcoutsos.github.io/keymap-editor/

ReFil, 2024-11-29:
Is anyone still having issues flashing the firmware from a mac? If not i'd like to close this issue out as solved in 24h

closed 2025-01-16 by KinesisCorporation

goDawgs18, 2025-01-17:
> <KinesisCorporation/Adv360-Pro-ZMK/issue/467/issue_event/15968799629@
> github.com>

I was never able to flash the device and ended up returning it.

lyleLH, 2025-02-25:
Sadly to post here. I just got my new 360Adv pro keyboard.

BUT, I only have two macbooks, I can not flash the firmware by dragging or cli cp/mv/rsync command.

So anyone addressed this issue? Feb 2025

KinesisCorporation, 2025-02-25:
> Sadly to post here. I just got my new 360Adv pro keyboard.
> 
> BUT, I only have two macbooks, I can not flash the firmware by dragging or cli cp/mv/rsync command.
> 
> So anyone addressed this issue? Feb 2025

Are you able to mount the drive?

What makes you think its not working?

macOS generates a file error but that can be ignored, the flash was still successful.

You can use the command Mod+V to print-to-screen the firmware version to confirm the update was successful.