KinesisCorporation/Adv360-Pro-ZMK #470: Macro is not working for copy and paste

ittechservice7github, 2024-05-10:
I use the action 1 key for RIGHT_COMMAND + C (copy) and action 3 for RIGHT_COMAND + V(paste).  When I go to use it, it just shows C on the screen.  

This is a fork from the update (KinesisCorporation#448) V3.

What am I doing wrong?

Below is the macro file and keyboard map file I have.

config/macros.dtsi:
  macro_quotes: macro_quotes {
    compatible = "zmk,behavior-macro";
    label = "macro_quotes";
    #binding-cells = <0>;
    bindings = <&kp SQT>, <&kp SQT>, <&kp LEFT>;
  };
  macro_dquotes: macro_dquotes {
    compatible = "zmk,behavior-macro";
    label = "macro_dquotes";
    #binding-cells = <0>;
    bindings = <&kp DQT>, <&kp DQT>, <&kp LEFT>;
  };
  macro_braces: macro_braces {
    compatible = "zmk,behavior-macro";
    label = "macro_braces";
    #binding-cells = <0>;
    bindings = <&kp LBRC>, <&kp RBRC>, <&kp LEFT>;
  };
  macro_parens: macro_parens {
    compatible = "zmk,behavior-macro";
    label = "macro_parens";
    #binding-cells = <0>;
    bindings = <&kp LPAR>, <&kp RPAR>, <&kp LEFT>;
  };
  macro_brackets: macro_brackets {
    compatible = "zmk,behavior-macro";
    label = "macro_brackets";
    #binding-cells = <0>;
    bindings = <&kp LBKT>, <&kp RBKT>, <&kp LEFT>;
  };
  macro_kinesis: macro_kinesis {
    compatible = "zmk,behavior-macro";
    label = "macro_kinesis";
    #binding-cells = <0>;
    bindings = <&kp LS(K)>, <&kp I>, <&kp N>, <&kp E>, <&kp S>, <&kp I>, <&kp S>;
  };
  macro_cmd_copy: macro_cmd_copy {
    compatible = "zmk,behavior-macro";
    label = "macro_cmd_copy";
    #binding-cells = <0>;
    bindings = <&kp RIGHT_COMMAND>, <&kp C>;
  };
  macro_cmd_paste: macro_cmd_paste {
    compatible = "zmk,behavior-macro";
    label = "macro_cmd_paste";
    #binding-cells = <0>;
    bindings = <&kp RIGHT_COMMAND>, <&kp V>;
  };

config/adv360.keymap
#include <behaviors.dtsi>
#include <dt-bindings/zmk/keys.h>
#include <dt-bindings/zmk/bt.h>
#include <dt-bindings/zmk/rgb.h>
#include <dt-bindings/zmk/backlight.h>

/ {
    behaviors {
      #include "macros.dtsi"
      #include "version.dtsi"
      #ifndef VERSION_MACRO
      macro_ver: macro_ver {
        compatible = "zmk,behavior-macro";
        label = "macro_version";
        #binding-cells = <0>;
        bindings = <&kp RET>;
      };
      #endif

      hm: homerow_mods {
          compatible = "zmk,behavior-hold-tap";
          label = "HOMEROW_MODS";
          #binding-cells = <2>;
          tapping-term-ms = <200>;
          quick_tap_ms = <175>;
          flavor = "tap-preferred";
          bindings = <&kp>, <&kp>;
      };
    };

  keymap {
    compatible = "zmk,keymap";

    default_layer {
      bindings = <
        &kp EQUAL &kp N1    &kp N2   &kp N3   &kp N4     &kp N5 &tog 1                                                           &mo 3 &kp N6 &kp N7 &kp N8    &kp N9   &kp N0   &kp MINUS
        &kp TAB   &kp Q     &kp W    &kp E    &kp R      &kp T  &macro_cmd_copy                                                            &macro_cmd_paste &kp Y  &kp U  &kp I     &kp O    &kp P    &kp BSLH
        &kp CAPS   &kp A     &kp S    &kp D    &kp F      &kp G  &kp LALT           &kp LGUI &kp LCTRL &kp RCTRL  &kp RIGHT_COMMAND           &kp RALT &kp H  &kp J  &kp K     &kp L    &kp SEMI &kp SQT
        &kp LSHFT &kp Z     &kp X    &kp C    &kp V      &kp B                            &kp HOME &kp PG_UP                           &kp N  &kp M  &kp COMMA &kp DOT  &kp FSLH &kp RSHFT
        &mo 2     &kp GRAVE &kp ESC &kp LEFT &kp RIGHT                &kp BSPC &kp DEL   &kp END  &kp PG_DN &kp ENTER &kp SPACE              &kp UP &kp DOWN  &kp LBKT &kp RBKT &mo 2
      >;
    };
    keypad {
      bindings = <
        &kp EQUAL &kp N1    &kp N2   &kp N3   &kp N4     &kp N5 &trans                                                           &mo 3 &kp N6 &kp KP_NUM &kp KP_EQUAL &kp KP_DIVIDE &kp KP_MULTIPLY &kp MINUS
        &kp TAB   &kp Q     &kp W    &kp E    &kp R      &kp T  &macro_cmd_copy                                                             &macro_cmd_paste &kp Y  &kp KP_N7  &kp KP_N8    &kp KP_N9     &kp KP_MINUS    &kp BSLH
        &kp CAPS   &kp A     &kp S    &kp D    &kp F      &kp G  &kp LALT           &kp LGUI &kp LCTRL &kp RCTRL  &kp RIGHT_COMMAND           &kp RALT &kp H  &kp KP_N4  &kp KP_N5    &kp KP_N6     &kp KP_PLUS     &kp SQT
        &kp LSHFT &kp Z     &kp X    &kp C    &kp V      &kp B                            &kp HOME &kp PG_UP                           &kp N  &kp KP_N1  &kp KP_N2    &kp KP_N3     &kp KP_ENTER    &kp RSHFT
        &mo 2     &kp GRAVE &kp ESC &kp LEFT &kp RIGHT                &kp BSPC &kp DEL   &kp END  &kp PG_DN &kp ENTER &kp KP_N0              &kp UP     &kp DOWN     &kp KP_DOT    &kp RBKT        &mo 2
      >;
    };
    fn {
      bindings = <
        &kp F1 &kp F2 &kp F3 &kp F4 &kp F5 &kp F6 &tog 1                                           &mo 3 &kp F7 &kp F8 &kp F9 &kp F10 &kp F11 &kp F12
        &trans &trans &trans &trans &trans &trans  &none                                           &none &trans &trans &trans &trans  &trans  &trans
        &trans &trans &trans &trans &trans &trans  &none        &trans &trans &trans &trans        &none &trans &trans &trans &trans  &trans  &trans
        &trans &trans &trans &trans &trans &trans                      &trans &trans                     &trans &trans &trans &trans  &trans  &trans
        &trans &trans &trans &trans &trans               &trans &trans &trans &trans &trans &trans              &trans &trans &trans  &trans  &trans
      >;
    };
    mod {
      bindings = <
        &none &bt BT_SEL 0 &bt BT_SEL 1 &bt BT_SEL 2 &bt BT_SEL 3 &bt BT_SEL 4 &none                                                                     &trans                 &none        &none        &none        &none &none &none
        &none &none        &none        &none        &none        &none        &bootloader                                                               &bootloader            &none        &none        &none        &none &none &none
        &none &none        &none        &none        &none        &none        &none                   &none &none &bt BT_CLR &none                      &rgb_ug RGB_MEFS_CMD 5 &none        &none        &none        &none &none &none
        &none &none        &none        &none        &macro_ver   &none                                      &none &none                                                        &none        &none        &none        &none &none &none
        &none &none        &none        &none        &none                                       &none &none &none &none      &bl BL_TOG &rgb_ug RGB_TOG                                     &bl BL_INC   &bl BL_DEC   &none &none &none
      >;
    };
  };
};

ittechservice7github, 2024-05-10:
Could someone please help me on how to get the macros to work?

ReFil, 2024-05-10:
Instead of using a macro for it you can do &kp LG(C) and &kp LG(V)

your macro isn't holding down left command whilst pressing C and V

ittechservice7github, 2024-05-11:
Thank you, I'll give that a try and update back.

ittechservice7github, 2024-05-11:
Thank you, @ReFil .  Your suggestion worked.  Appreciate your help.

closed 2024-05-11 by ittechservice7github