KinesisCorporation/Adv360-Pro-ZMK #471: Is the firmware updated on the keyboard?

ittechservice7github, 2024-05-10:
I used a fork from V3 [https://github.com/KinesisCorporation/Adv360-Pro-ZMK/pull/448[)]

I remapped a few keys around.  

Every time I load the firmware, I get the following error:
"The Finder can't complete the operation because some data in "left.uf2" can't be read or written. (Error code -36)"

When I check the INFO_UF2.TXT, I says 
"UF2 Bootloader 0.4.0-41-g103bd32-dirty lib/nrfx (v2.0.0) lib/tinyusb (0.9.0-22-g7cdeed54) lib/uf2 (remotes/origin/configupdate-9-gadbb8c7)
Model: ADV360PRO
Board-ID: nRF52840-ADV360PRO-v1
SoftDevice: S140 version 6.1.1
Date: Jun 29 2022
"

All the files' last mod date is Jun 29, 2p22 11:03 PM.

From the clues, and error. I don't think the firmware get updated.

I load the keyboard using the paper clip hold for firmware flashing.

Thanks for your help.
[image]

ReFil, 2024-05-10:
This error is expected, you can use mod+V to check the version running on the board, it will type out the timestamp and git commit

ittechservice7github, 2024-05-10:
Thank you, Refil.

I did mod+V.  I guess it is loaded fine.  The last modified date time and info in the INFO file throw me off. Thank you.
20240510-v3.0-749bba1

closed 2024-05-10 by ittechservice7github